JacKeTUs/universal-pidff #1: Damping/friction/spring effects don't work in games

Lawstorant, 2024-05-27:
Official issue to track progress.

These effects don't seem to be working when playing games/using WheelCheck.exe I can't see anything out of ordinary in dmesg logs when these affect are triggered.

For some reason, playing "conditional" example for ffbplay from https://github.com/berarma/ffbtools triggers damping/friction/spring correctly.

Lawstorant, 2024-05-28:
Update

I think I found the culprit for this weird behavior. Moza FW seems to care about the direction for the Spring/damper/friction effects. I don't know that's the direction value sesnt by games and `WheelCheck.exe` yet, but after playing around with `ffbplay` in interactive mode I got these results:
- Direction range can be set **0-65535** (it seems like this range changes direction clockwise)
- values 0, 32768, 65535 give absolutely no force. 
- values **16384 and 49152** have the strongest force
- value **16384** gives us the proper spring/damper/friction effects
- value 49152 reverses those effects

I'll try hardcoding 16384 as the direction in `hid-pidff.c` for now to see If it'll work for games and `WheelCheck.exe`. Afterwards, we can decide on how to proceed with this direction stuff as it would be best to maybe try to upstream any fixes to pidff (if we manage a clean solution)

Lawstorant, 2024-05-28:
Okay, I got the effects to always work by hardcoding the direction value to 16384. It's an ugly workaround, but I don't have a better one for now.

Lawstorant, 2024-05-28:
I introduced quirks to hid-pidff and fixed the issue: https://github.com/JacKeTUs/moza-ff/pull/2

JacKeTUs, 2024-05-29:
For reference
https://www.usb.org/sites/default/files/documents/pid1_01.pdf
Section 4.1, 5.3

> If there is a single Condition report block for an effect with more than one axis, then the direction along
which the parameters of the Condition report block are in effect is determined by the direction parameters
passed in the Direction field of the Effect report block. For example, a friction condition rotated 45
degrees (in polar coordinates) would resist joystick motion in the northeast-southwest direction but would
have no effect on joystick motion in the northwest-southeast direction.

JacKeTUs, 2024-05-29:
Will look into that today

Lawstorant, 2024-06-02:
Resolved after merging PR

closed 2024-06-02 by Lawstorant