JacKeTUs/universal-pidff #4: Will this package release for other distros?

ID86s, 2024-06-09:
i can't find it for gentoo as emerge package, will it release for other packages managers?

Lawstorant, 2024-07-01:
Only if someone using Gentoo will actually package it. This is a community-driven endeavor. I only use Arch so I'm maintaining an AUR package.

r-burns, 2024-07-13:
Hi! Just wanted to say I'm happily using this on NixOS. Thanks for your work on this! I packaged it up like this:

<details>

```nix
{
  stdenv,
  lib,
  fetchFromGitHub,
  kernel,
}:

stdenv.mkDerivation {
  pname = "moza-ff";
  version = "unstable-2024-05-29";

  src = fetchFromGitHub {
    owner = "JacKeTUs";
    repo = "moza-ff";
    rev = "1ac6938ed4f553183950f12f89f6397f4cf4f605";
    hash = "sha256-XPg4VhYHFAUJF+zGO/Uj676Pjbp1L2N1rldxw8wLaAA=";
  };

  nativeBuildInputs = kernel.moduleBuildDependencies;

  postPatch = ''
    sed -i 's|depmod|#depmod|' Makefile
  '';

  makeFlags = kernel.makeFlags ++ [
    "KDIR=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"
  ];

  installFlags = [
    "INSTALL_MOD_PATH=${placeholder "out"}"
  ];

  meta = with lib; {
    description = "Force feedback support for MOZA steering wheel";
    homepage = "https://github.com/JacKeTUs/moza-ff";
    license = licenses.gpl2Plus;
    maintainers = [ maintainers.r-burns ];
    platforms = platforms.linux;
  };
}
```

</details>

I'm a nixpkgs committer and I'd be happy to contribute this and maintain it if any other NixOS users want to use the module.

JacKeTUs, 2024-07-15:
Hello! Just to let you know: we think "universal-ff" will better suite the purpose of bringing direct-drive wheels to the life on Linux
Please, update homepage/project name/repo URL to it. Thanks!

Lawstorant, 2024-08-09:
@DaRacci I recommend just using the Flatpak version available on [Flathub](https://flathub.org/apps/io.github.lawstorant.boxflat). You only need to add one udev rule manually and you're good to go, without the need to package and maintain anything. I myself use it and only maintain the AUR package as it's trivial.

danerieber, 2024-08-09:
@r-burns @DaRacci  I am also on NixOS and I packaged both universal-pidff and boxflat in [this repo](https://github.com/danerieber/moza-racing-wheel-nix)

DaRacci, 2024-08-10:
> @DaRacci I recommend just using the Flatpak version available on [Flathub](https://flathub.org/apps/io.github.lawstorant.boxflat). You only need to add one udev rule manually and you're good to go, [...]

While i could add Flatpak support for my System, I currently don't have it nor do i want to enable Flatpak, its just a matter of preference but i would much rather have a native build.

@danerieber If you are up for it, it may be worth upstreaming this into nixpkgs, if you end up doing this I'd be happy to help maintain it.

danerieber, 2024-08-10:
> @danerieber If you are up for it, it may be worth upstreaming this into nixpkgs, if you end up doing this I'd be happy to help maintain it.

@DaRacci I made a fork with the package [here](https://github.com/danerieber/nixpkgs/tree/universal-pidff). I don't have enough time to properly test it right now, and I also need to sit down and read the PR guidelines for nixpkgs, but it would be awesome if you could test it on your hardware and lmk if it works!

DaRacci, 2024-08-11:
@danerieber I don't actually have a moza wheel to test with yet, I've got one on a preorder and I've just been prepping for it, but the kernel compiles and loads with the module correctly, I've used nixpkgs-review to test it too.

danerieber, 2024-08-13:
@DaRacci submitted! https://github.com/NixOS/nixpkgs/pull/334445

JacKeTUs, 2024-10-17:
Hey guys, are we happy with NixOS packaging? Do we need to wait for someone implementing package for Gentoo emerge, or we can close this issue? :D

DaRacci, 2024-10-18:
NixOS Side is happy, however it may be worth converting into a discussion for future Distro packaging talks.