JacKeTUs/universal-pidff #7: Moza R5 Not detected/No forced feedback

cjhelloletsgo, 2024-07-14:
I am trying to get my Moza R5 working but it is not detected by steam, I think it should detected be since it is on the sdl whitelist now right? I also tried this suggestion in the README

Also, recent updates to SDL created SDL Hint variable to dynamically extend wheel devices list[17](https://github.com/JacKeTUs/linux-steering-wheels#user-content-fn-15-52e7281ea0014cd2f1ab0f06c7e837f6). You need to just export SDL_JOYSTICK_WHEEL_DEVICES=0x/0x,0x/0x before you launching something.

so for this device: Bus 001 Device 014: ID 346e:0004 Gudsen MOZA R5 Base

I ran this command export SDL_JOYSTICK_WHEEL_DEVICES=0x346e/0x0004
with no effect.

If I disable steam input then go into Dirt Rally 2.0 it detects Gudsen MOZA R5 Base but there is no forced feedback.

I also tried with the universal-quirks branch with the same effect.

spikerguy, 2024-07-14:
Can you try to add `SDL_JOYSTICK_WHEEL_DEVICES=0x346e/0x0004 %command%` in steam game command and launch the game?

If that doesn't help then try to launch control panel for that game 
`protontricks -c "wine control" 690790` and check if you can see your wheel in dinput or xinput. You have to see it in dinput so you can press override.

cjhelloletsgo, 2024-07-14:
I tried the first command and it was the same. 

I haven't used protontricks before but since I am on Ubuntu 24.04 I installed the flatpak version as they recommended.
Then I ran this command
```bash
flatpak run com.github.Matoking.protontricks -l
```

and found the app id that you gave me 

DiRT Rally 2.0 (690790)

So i ran this command:

```bash
flatpak run com.github.Matoking.protontricks -c "wine control" 690790
```

which gave me this error:
```
Traceback (most recent call last):
  File "/app/bin/protontricks", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/cli/main.py", line 32, in cli
    main(args)
  File "/app/lib/python3.11/site-packages/protontricks/cli/util.py", line 159, in wrapper
    return cli_func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/cli/main.py", line 339, in main
    proton_app = find_proton_app(
                 ^^^^^^^^^^^^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/steam.py", line 848, in find_proton_app
    tool_app = find_steam_compat_tool_app(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/steam.py", line 609, in find_steam_compat_tool_app
    appinfo_sections = [
                       ^
  File "/app/lib/python3.11/site-packages/protontricks/steam.py", line 609, in <listcomp>
    appinfo_sections = [
                       ^
  File "/app/lib/python3.11/site-packages/protontricks/steam.py", line 531, in iter_appinfo_sections
    raise SyntaxError(
SyntaxError: Invalid file magic number. The appinfo.vdf version might not be supported by the current version of Protontricks - please check for updates.
```

It looks like there is a bug causing it https://github.com/Matoking/protontricks/issues/304

But setting the proton version as an environmental variable or using the Steam stable branch rather than beta just gives me this:
```
pressure-vessel-wrap[9]: E: linkat: No such file or directory
Traceback (most recent call last):
  File "/app/bin/protontricks", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/cli/main.py", line 32, in cli
    main(args)
  File "/app/lib/python3.11/site-packages/protontricks/cli/util.py", line 159, in wrapper
    return cli_func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/cli/main.py", line 381, in main
    returncode = run_command(
                 ^^^^^^^^^^^^
  File "/app/lib/python3.11/site-packages/protontricks/util.py", line 516, in run_command
    raise RuntimeError(
RuntimeError: bwrap launcher crashed, returncode: 1 
```

Lawstorant, 2024-07-14:
This is am issue with codemasters games. They need EXPLICIT device mentions in their files, which is completely stupid. Look for "Dirt Rally moza fix" on youtube/steam pages.

cjhelloletsgo, 2024-07-15:
Are you referencing manually inputting the deviceinto the device_defines.xml file like in this reddit post? https://www.reddit.com/r/EASPORTSWRC/comments/17kd4zl/totally_disappointed/

I added it and still nothing

JacKeTUs, 2024-07-15:
Can you try different game? For example, ETS/ATS? To test if driver works correctly.
Or tests from [berarma/ffbtools](https://github.com/berarma/ffbtools) ?

Lawstorant, 2024-07-15:
I own both Dirt Rally and Moza R9. Let me test it myself

Lawstorant, 2024-07-15:
Yup, FFB works. Proton 9.0-2 and no file fixes necessary. Just use DirectInput wheel preset, set all the input bindings, calibrate your wheel and it should work. Sadly, DR seems to forget these settings after you closed the game.

Honestly, I think there's just no need to play it anymore? Dirt  Rally 2.0 has all the content from DR and is less finicky about this stuff.

Lawstorant, 2024-07-15:
For Dirt Rally 1 you don't need to do anything
For Dirt Rally 2:
```
<?xml version="1.0" encoding="utf-8"?>
<device_list>
  <!--
    Device configuration file.
    This file is used to define device definitions and their usage in the game.

    To add a new device add a device node with following attributes:
      id -  The hardware id for PC devices is essentially '{PID#VID#-0000-0000-0000-504944564944}' where PID# and VID# is replaced with the Product ID and Vender ID of the device.
            You can get these ID's by looking in Devices & Printers, Properties on the required device, Hardware tab, USB Input Device, Properties. In 'Information' you should see VID_#### and PID_####. 
            Take the hex values proceeding the underscore and replace into the above sample.
            e.g. 
              'Device USB\VID_045E&PID_028E&IG_00\8&3afa0edf&0&00 was configured.'
              {028E045E-0000-0000-0000-504944564944}
      name - The name of the device matching the name used in the action maps.
      priority - The higher the priority the more likely the device is used in automatic selection mode as the steering device over other devices.
      {
        0 - reserved for not_supported devices
        1-5, 99 - reserved for library defaults
        6-98 - pads
        100+ - wheels
      }
      version (Optional) - Hardware version of the device, used in conjunction with action map device versioning
      type (Optional) - The type of device, if set to 'unknown' will use the type returned from the driver, otherwise it will overwrite with the selection.
      {
        wheel
        pad
        joystick
        keyboard
        mouse
        pedal
        shifter
        handbreak
        unknown - Default
        not_supported
      }
  
      ffb (Optional)
      {
        default - Default - Uses the value returned from the drivers
        enabled - Force enables the ffb even if the drivers return disabled.
        disabled - Force disables the ffb even if the drivers return enabled.
      }
      ffb_force (Optional) - 0.66 is the lowest value to enable user to scale back up to 1.
      {
        1.0 - Default
      }
      ffb_friction (Optional)
      {
        1.0 - Default
      }
      ffb_high_pass (Optional)
      {
        0.0 - Default
      }
      default (Optional)
      {
        false - Default
        true - Used only by input library defaults
      }
      collections_max (Optional) -  A collection limit >1 or 0 will allow new devices to be created with the postfix '_col##' to the hardware id.
                                    These devices can be used to extend functionality with additional device id's if the device supports them.
                                    e.g.
                                      <device id="{00050EB7-0000-0000-0000-504944564944}" name="ftec_csl_p1" priority="100" type="wheel" collections_max="2" />
                                    
                                    You can now setup an optional collection id to give the second collection it's own action map.                                    
                                      <device id="{00050EB7-0000-0000-0000-504944564944}_col02" name="ftec_csl_p1_col02" priority="100" type="wheel" ffb="disabled" />
      {
[... 67 lines cut ...]
  <device id="{B695044F-0000-0000-0000-504944564944}" name="tm_tss_hb" priority="100" type="handbrake" />
  <device id="{B692044F-0000-0000-0000-504944564944}" name="tm_ts_xw" priority="100" type="wheel" />
  <device id="{B684044F-0000-0000-0000-504944564944}" name="tm_t_gt" priority="100" type="wheel" />
  <device id="{B681044F-0000-0000-0000-504944564944}" name="tm_t_gt" priority="100" type="not_supported" />
  <device id="{62040EB7-0000-0000-0000-504944564944}" name="ftec_csl_pedals" priority="100" type="pedal" />
  <device id="{07051532-0000-0000-0000-504944564944}" name="rz_raiju_mobile_pad" priority="94" type="pad" default="true" />
  <device id="{00060EB7-0000-0000-0000-504944564944}" name="ftec_dd_f1_pc" priority="100" type="wheel" />
  <device id="{00070EB7-0000-0000-0000-504944564944}" name="ftec_dd_2_pc" priority="100" type="wheel" />
  <device id="{0FFBF055-0000-0000-0000-504944564944}" name="simucube_osw_mmos" priority="100" type="wheel" official="false" />
  <device id="{0D5A16D0-0000-0000-0000-504944564944}" name="simucube_osw" priority="100" type="wheel" official="false" />
  <device id="{0D6116D0-0000-0000-0000-504944564944}" name="simucube2_sport" priority="100" type="wheel" official="false" />
  <device id="{0D6016D0-0000-0000-0000-504944564944}" name="simucube2_pro" priority="100" type="wheel" official="false" />
  <device id="{0D5F16D0-0000-0000-0000-504944564944}" name="simucube2_ultimate" priority="100" type="wheel" official="false" />
  <device id="{09CC054C-0000-0000-0000-504944564944}" name="ps4_pad" priority="100" type="pad" />
  <device id="{C26E046D-0000-0000-0000-504944564944}" name="lg_g923_xone" priority="100" type="wheel" />
  <device id="{C266046D-0000-0000-0000-504944564944}" name="lg_g923_ps4" priority="100" type="wheel" />

  <device official="false" name="moza_wheel_r16_r21" type="wheel" id="{0000346E-0000-0000-0000-504944564944}" priority="100"/>
  <device official="false" name="moza_wheel_r16_r21" type="wheel" id="{0000346E-0000-0000-0000-504944564944}" priority="100"/>
  <device official="false" name="moza_wheel_r9" type="wheel" id="{0002346E-0000-0000-0000-504944564944}" priority="100"/>
  <device official="false" name="moza_wheel_r5" type="wheel" id="{0004346E-0000-0000-0000-504944564944}" priority="100"/>
  <device official="false" name="moza_wheel_r3" type="wheel" id="{0005346E-0000-0000-0000-504944564944}" priority="100"/>
  <device official="false" name="moza_wheel_r12" type="wheel" id="{0006346E-0000-0000-0000-504944564944}" priority="100"/>
</device_list>
```

cjhelloletsgo, 2024-07-16:
I started up boxflat and it says my forced feedback is at 80%.
[image]

Steam input is disabled
[image]

I tried @Lawstorant's xml file, it looks the same as mine and didn't make a difference in game.

it still is showing up but no forced feedback :(
It is set as a wheel instead of a controller or anything else
[image]

And forced feedback override is set to Device Driver
[image]

I am on proton experimental but tried with proton 8.0-5 and 9.0-2 and it made no difference.

As for this: 
Can you try different game? For example, ETS/ATS? To test if driver works correctly.
I don't have any other racing games yet to try out

Or tests from [berarma/ffbtools](https://github.com/berarma/ffbtools) ?
I looked at this but I honestly don't understand how to use it.

Sorry I'm being a pain here lol, I appreciate your help.

Lawstorant, 2024-07-16:
No worries! We all had the same problems at some point.
I did some updates to boxflat, so you shouldn't see these errors anymore. And there's a .desktop file so you should be able to easily find it in your applications.

For the FFB part, I've run this driver with: Dirt rally, Dirat rally 2.0, EA WRC, WRC 10, ETS/ATS, ACC.
And yeah, I was doing extensive testing with ffbtools/ffbplay when I was ding quirks for Moza.

May I ask, how did you install your driver?

JacKeTUs, 2024-07-16:
@Lawstorant, i think @cjhelloletsgo need to wait a bit, we need to merge quirks fix :D

cjhelloletsgo, 2024-07-16:
I installed using the dkms method in the README, here is a screenshot to confirm it is installed
[image]

[here](https://github.com/user-attachments/files/16257343/dmesg.txt) is the output of `sudo dmesg`

I gave [berarma/ffbtools](https://github.com/berarma/ffbtools) another try and got it to compile at least but I still don't really get how I am supposed to use it.

For example I ran `make` and it built
[image]

but if I actually try to run interactively it can't find the device
[image]

This is the doc page, all it says is device so I don't really know what it needs passed in.
[image]

JacKeTUs, 2024-07-17:
First, can you please update `universal-pidff` to latest version?
Second, can you please check in mobile Moza app that you have wheelbase power limit is not 0?
Commands you ran should output *some* FFB, but if nothing happens, may be something wrong with your wheelbase?... Is it working on Windows? What firmware are you running?

JacKeTUs, 2024-07-17:
Hmm, at second glance, i see wrong message in your dmesg, from stock pidff driver, not from universal-pidff
Do you have SecureBoot enabled? May be DKMS can't load module properly? 
Can you try manually load the driver with `insmod`?

cjhelloletsgo, 2024-07-17:
IT WORKS!!!

I think you were right on the SecureBoot stuff. I ended up doing this:

After step 3 in the manual build steps in the README I generated a key pair:
```
openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Your Name/"
```

Enrolled the key:
```
sudo mokutil --import MOK.der
```

Rebooted and was greeted with this screen:
[image]

Selected the "Enroll MOK" option and added the key by confirming the password

Signed the module:
```
sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 MOK.priv MOK.der hid-universal-pidff.ko
```

Loaded the module:
```
sudo insmod hid-universal-pidff.ko
```

Confirmed the module was loaded:
```
sudo dmesg | grep hid-universal-pidff
```
[image]

Then tested again with:
```
./build/ffbplay -d /dev/input/event14 -t tests/sine.ffb
```

and finally got the forced feedback. Also tested with Dirt Rally 2.0 and it works in there as well. Thanks for the help everyone!

closed 2024-07-17 by cjhelloletsgo

JacKeTUs, 2024-07-17:
I will add these steps for signing module on SecureBoot system to the docs, thank you for the follow-up and enjoy racing :D

JacKeTUs, 2024-07-17:
@cjhelloletsgo, please check instructions in branch `signing` :pray: 
https://github.com/JacKeTUs/universal-pidff/pull/15

Teteros, 2024-12-27:
The thread was very useful, thank you all.
I'd like to add that I needed to toggle `Work mode` in boxflat to get any FFB on my Moza R5, (it was on and i set it off then on) this made ffbplay work.

finding id was easiest by using the by-id subdir:
`ffbplay -d /dev/input/by-id/usb-Gudsen_MOZA_R5_Base*-event-joystick -t sine.ffb`

JacKeTUs, 2024-12-27:
Hmm, @Lawstorant, what is "Work Mode" exactly? :D 
Is this value non-persistent across wheelbase poweroffs?

Lawstorant, 2024-12-27:
This is a bad name to basically mean "soft power on/off". It cuts power from the FFB motor AND the connected wheel. There's a similar toggle to enable/disable FFB by the `e-stop` but it doesn't cut power.

This setting is persistent like 95% of them, but they sometimes don't reflect the actual state (they get stuck). So yeah, flipping them off and on again fixes things.