JacKeTUs/universal-pidff #18: FFB Lacking Detail in A-B Comparissons with Windows [Moza R5 & Ubuntu 22.04]

Done-In-60-2nds, 2024-08-04:
In A-B comparisons to game-play under windows, I've noticed that there is significantly greater detail in force feedback effects under Windows, even when using the same in-game and base settings. Is this normal? Is there currently any way to troubleshoot this issue?

Lawstorant, 2024-08-05:
There may be some differences du to how pidff currently handles envelopes. Other than that, it could be that the translation from windows is not perfect.

I did some work on envelopes to make the FFB better and for it to work with games like rFactor 2 ere: https://github.com/JacKeTUs/universal-pidff/pull/16

Lawstorant, 2024-08-05:
@Done-In-60-2nds 
Could you list out the games you tested? I may be able to compare them as well. Apart from the envelope stuff, I compared the HID data that's sent on windows and linux to the wheelbase and it's identical in 99% of cases.

JacKeTUs, 2024-08-08:
@Done-In-60-2nds, can you please check if new version will fix effects for you?

JacKeTUs, 2024-08-08:
@Lawstorant, we probably should write some tests for ffbplay (https://github.com/berarma/ffbtools) to see if every effect/combination will work with our changes

Lawstorant, 2024-08-08:
Sure, I could do it as the included sine test is a good starting point (it has constant force emulation test). Just have to add other types of effects and maybe make the transitions faster, as 5 seconds is just a bit too long.

Lawstorant, 2024-08-16:
`effect-test.ffb` for `ffbplay` is now available in the main branch.
@Done-In-60-2nds could you give it a try? We don't be able to help if you won't provide additional info.

Lawstorant, 2024-09-22:
@JacKeTUs maybe this should be closed as we didn't hear back in over a month?

JacKeTUs, 2024-09-22:
Honestly, I'm against "closed because of no response" thing. Issue *was* there, maybe somebody too could verify it / check it

DTRGator, 2024-10-31:
I can A-B test it both under Linux and & Windows, I could only possibly test it with Forza Horizon 5. I will update you guys on the results.

JacKeTUs, 2024-10-31:
If you could test it, it would be great!
Also, if you notice that something off, could you, please, record USB traffic with Wireshark on both systems with ~similar conditions? Like, maybe, dumper is more stronger/weaker on one system, then we could look into .pcap files and see if game/wine outputs different commands/gain, etc...

DTRGator, 2024-11-01:
So a small update, 
Since I’m not entirely sure what a “dumper” is (I apologize for my lack of familiarity with these terms; I should’ve done more research before posting), I left all values in Boxflat at their default settings, and the in-game settings are also on default.

To keep the testing consistent, I plan to run the same event on both Windows and Linux. I was able to capture a .pcap file for Linux (with lsusb being a huge help in identifying the Wheelbase connection). However, on Windows, I'm having trouble identifying which USB bus the Wheelbase is connected to. This would allow me to filter out other devices and reduce noise in the communication data between the Wheelbase and the host, which would make it easier for you to identify.

I am at my wits end on identifying the USB bus connection for the Wheelbase in Windows.  Is there a different approach I should take ? I’d appreciate any pointers.

JacKeTUs, 2024-11-01:
Damper - Force against speed of the wheel, wheel is harder to turn in general, and it de facto defines maximum speed you can turn the wheel with
Inertia - Force against acceleration of the wheel, wheel is slightly harder to turn at the beginning of the motion
Friction - Force against any movement of the wheel. Virtual effect is very hard to notice, because every (and i mean every) wheelbase have *some* friction because of the nature. 

Spring - Force value is a proportion of distance from the center (or offset point) - wheel always trying to return to the center / offset point. 
Constant Force - whatever force game outputs. It's the main way to games output whatever they want. 
Sin Force - Force value is defined by sin(t) function, where t - is the time. There is also coefficients, period, etcetc. Game can use it for simulating "engine rumbling"
Square Force, Triangle Force - Force value is defined by square form or triangle form (literally). May be used for simulating crashes/bumps. (at certain time wheel suddenly turns left and then back - square, or slowly turns left and then suddenly back - triangle)

Games usually just uses Constant Forces and Dampers/Inertia. Sin/Square/Triangles are not used so much. Constant Forces are updated by games at high frequency, and Dampers/Inertia/Friction usually just initialized at the beginning of 'driving', and stays there until you change the car/change the track.

JacKeTUs, 2024-11-01:
For USB identification in Windows i can only suggest starting recording before connecting wheelbase to the PC, and then watch Wireshark window for new usb devices... :(

Lawstorant, 2024-11-01:
Heh, as I spent countless hours with Moza and Wireshark, you'd want to look for URB_INTERRUPT packets with data starting with `7e`. This indicates serial messages from PitHouse. HID data will be available on the same bus.

But to seve you the trouble, I know that the data from games to the wheelbase is exactly the same on Windows and Linux, now that we're fixing the direction to `0x4000 (9000)`. That's why I doubted the OP in the first place. With My R9 V2, I can't feel any difference between the two OSes.

Oh, and Forza Horizon might not be the best title to test FFB. Well, it might be one of the worst :P

DTRGator, 2024-11-01:
@Lawstorant If that’s indeed the case, capturing USB traffic for this particular game might not provide useful insights. However, I’m still noticing a difference in detail on Linux compared to Windows, which could suggest FH5 (or possibly the Wine calls) as the source of the issue. Do you still want me to test Forza regardless?

To verify, I’ll test with Assetto Corsa to see if any differences appear between the two operating systems. Thanks again for the insights, @JacKeTUs and @Lawstorant!

Lawstorant, 2024-11-01:
I did majority of testing with EA WRC and GT Legends/rFactor 2. The latter two use sine efdect to emulate constant force

DTRGator, 2024-11-10:
Apologies, got bogged down with exams and the like, I did test it nonetheless with Windows and Linux on Forza Horizon 5 using the default settings and the same in game events (I did this since I found the biggest difference in FFB detail in this game.) 

.pacp files Linux : [https://drive.proton.me/urls/WG24XXP95R#6NHPDR2z4kte](url)
.pacp files Windows : [https://drive.proton.me/urls/XBTA44CGTR#77D3OZvpjQcE](url)

Wheelbase on windows `usb.src == 2.10.3 or usb.dst == 2.10.3`
Wheelbase on Linux `usb.src == 1.6.3 or usb.dst == 1.6.3`

Lawstorant, 2024-11-10:
I'd say the issue there is that on Windows, Pit House is injecting a `.dll` file into Horizon to make the FFB better. It's still quite bad, but feels a bit better than stock. You could try fishing out this `.dll.` file and adding it on linux as well.

Lawstorant, 2024-11-10:
Instead, you can use something like `wheelcheck.exe` and see for yourself, that the values are identical.