JacKeTUs/universal-pidff #29: Moza R5 No FFB

OPNightMarine, 2024-08-24:
Tried both, DKMS and manual install, no FFB on either. Using Boxflat.

System - 

OS: Pop!_OS jammy 22.04 x86_64
Kernel: Linux 6.9.3-76060903-generic

I don't know what other info would help, would love to be able to race on linux.
Much appreciated.

cjhelloletsgo, 2024-08-24:
Did you follow the steps in the signing section? I had a similar problem and that fixed it for me. https://github.com/JacKeTUs/universal-pidff/blob/main/docs/SIGNING.md#signing

Lawstorant, 2024-08-24:
Any errors related to loading unsigned module? Are you running your OS with Secure Boot enabled?

OPNightMarine, 2024-08-24:
> Did you follow the steps in the signing section? I had a similar problem and that fixed it for me. https://github.com/JacKeTUs/universal-pidff/blob/main/docs/SIGNING.md#signing

I missed this part in the docs, hopefully this is the reason.

OPNightMarine, 2024-08-24:
> Any errors related to loading unsigned module? Are you running your OS with Secure Boot enabled?

I don't have secure boot enabled, from what i can tell i get no errors, just no FFB.

cjhelloletsgo, 2024-08-24:
> > Any errors related to loading unsigned module? Are you running your OS with Secure Boot enabled?
> 
> I don't have secure boot enabled, from what i can tell i get no errors, just no FFB.

Secureboot is enabled by default on Ubuntu. Did you explicitly disable it? If you are sure that it is not enabled then ignore this. 

I would install mokutil 

```
sudo apt install mokutil -y
```
 
Then run 
``` 
sudo mokutil --sb-state​
```

To confirm secureboot status.

OPNightMarine, 2024-08-24:
Yup I made sure it's disabled, while going through the signing docs i noticed it's for secure boot so i assume I skip that part. While looking around, I found the location for DKMS folder that holds the files and found out this in the make log.

make[1]: Entering directory '/usr/src/linux-headers-6.9.3-76060903-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0

I have no experience on how to use the kernel I have to compile.

Lawstorant, 2024-08-24:
So the issue is multifaceted but not caused by universal-pidff at all.

1. There's a compiler mismatch. You should update your kernel of you can. Run apt update and then full-upgrade.
2. Pop OS is another broken piece of... software from System 76. They're still using ubuntu 22.04 ad the base. Heah, it's more than 2 hears old at this point and we already have a new LTS with 24.04.

Nothing we can do about it. You should do yourself a favor and switch to vanilla ubuntu, or beyter yet, to something like fedora. Gaming-focused distros don't make any sense. Arch might not be for you yet, but Cachy OS seems to be an easy way in.

TL;DR old packages bad, Pop OS bad, System76 bad.

OPNightMarine, 2024-08-24:
> So the issue is multifaceted but not caused by universal-pidff at all.
>
> 1. There's a compiler mismatch. You should update your kernel of you can. Run apt update and then full-upgrade.
> 2. Pop OS is [...]

Kernel is already latest, Issue is just the compiling, which i don't have much experience in. I've looked around to see how i can compile using the same, haven't been able too find it yet. The only difference I can see is the main kernel used x86_64 to compile while it wasn't used for the compiling of universal-pidff.  You have your own option about pop os. It's the one that works very well and stable on my system and there's nothing wrong with 22.04 compared to 24.04 especially considering xwayland problems. The "nothing you can do" part i say is not true, this is linux.

OPNightMarine, 2024-08-25:
Well I don't know what changed, the FFB is now working, even with the compiler warning. I'll close the issue. Now It's time to go racing.

Much appreciated everyone.

closed 2024-08-25 by OPNightMarine