JacKeTUs/universal-pidff #31: Moza R5 no FFB dmesg spam

WaneTol72, 2024-08-26:
dmesg
```
[ 2008.582758] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.582816] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.582872] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.582927] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.582982] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583036] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583093] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583148] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583203] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583258] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583314] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583426] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583481] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583536] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583592] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583647] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583701] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583756] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583811] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583865] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583920] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.583976] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584030] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584085] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584141] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584196] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584252] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584307] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584416] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584471] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584526] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.584584] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695570] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695628] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695685] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695771] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695828] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695883] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695940] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.695996] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696051] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696106] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696163] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696219] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696274] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696335] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696468] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696525] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696580] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696636] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696692] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696750] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696806] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696861] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696917] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.696972] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697028] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697093] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697151] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697206] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697263] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697319] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697429] hid-universal-pidff 0003:346E:0004.0009: output queue full
[ 2008.697485] hid-universal-pidff 0003:346E:0004.0009: output queue full
```

Lawstorant, 2024-08-27:
This could be related to FFB command spam. Could you try and lower the FFB refresh rate? Below 500 Hz would be great

JacKeTUs, 2024-08-28:
Yep, Full update type and Slow FFB reports will fix the issue for BeamNG

WaneTol72, 2024-08-28:
I solved the problem in another way - I ran the native Linux version of BeamNG. Everything worked out of the box.

I apologize if I confused - before that I was running the game via ProtonGE 9.11. I will try to play this version now and apply settings according to your advice, it will help others to find the answer to a similar problem!

WaneTol72, 2024-08-28:
Yes, changing the FFB refresh rate parameter fixes the problem. By the way, it doesn't matter which FFB refresh rate parameter I choose - it works on any parameter, most likely BeamNG can't detect the frequency when first initializing the base. \

I think we can close the issue, thanks!

Lawstorant, 2024-08-28:
Nah, moza just handles 1000 Hz without issues as it uses plain HID PID commands.

closed 2024-08-28 by JacKeTUs