JacKeTUs/universal-pidff #32: No Forced Feedback

cjhelloletsgo, 2024-09-22:
Previously forced feedback was working for me with my Moza R5 but it has stopped working.

I think it may have been after a kernel update? When I discovered it wasn't working I uninstalled per the README then followed the instructions for installation again with the DKMS method. 

After reboot the module is loaded
[image]

I tried to confirm that forced feedback was working via the test file but it doesn't work for me.
[image]

No forced feedback in games either.

Lawstorant, 2024-09-22:
Please provide more info like your Distribution and it's version

cjhelloletsgo, 2024-09-23:
This was my bad, when I ran ```sudo dmesg``` there was an error with signing. I realize now that is because I updated my motherboard firmware and that clears the keys. Re-enrolled the key and it works again. Thank You!

closed 2024-09-23 by cjhelloletsgo