JacKeTUs/universal-pidff #33: Moza R5 no longer recognized after restarting

ollipop-ops, 2024-09-26:
My Moza R5 was working fine with this driver and Boxflat but today it no longer works, I can't see any messages in dmesg and Boxflat doesn't even recognize that I have a wheel plugged in now.
I'm on LMDE6 if that helps

Lawstorant, 2024-10-01:
Please provide output of `lsusb`, `ls /dev/serial/by-id/` and `ls /dev/input/by-id`

There haven't been any recent changes to this driver so the issue must be something i  your system.

ollipop-ops, 2024-10-01:
I rebuilt and reinstalled the kernel module to the same issue. Weirdly enough what ended up resolving the issue was switching the USB port that my Moza R5 was plugged into and now it's working consistently. Very odd as I haven't had issues with that USB port before or since

Lawstorant, 2024-10-01:
Computers right? Haha

Lawstorant, 2024-10-17:
@JacKeTUs can be closed

closed 2024-10-17 by JacKeTUs