JacKeTUs/universal-pidff #34: Moza R3 Racing Wheel and Pedals not detected

danhultg, 2024-09-27:
I'm trying to use this module to hook up the Moza R3 Racing Wheel and Pedals base but it's not getting detected correctly. I see no mention of universal-pidff in dmesg when connecting it. Here is the dmesg output after connecting:

```
[  438.606315] usb 1-1.3: new full-speed USB device number 10 using dwc_otg
[  438.767211] usb 1-1.3: New USB device found, idVendor=346e, idProduct=0017, bcdDevice= 1.00
[  438.767242] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  438.767257] usb 1-1.3: Product: R3 Racing Wheel and Pedals
[  438.767270] usb 1-1.3: Manufacturer: Gudsen
[  438.767281] usb 1-1.3: SerialNumber: 320038000851303335333433
```

I have tried installing this both using the dkms approach, and manual build followed by insmod, but still no success.

I noticed that the idProduct=0017 was not included in the hid-ids.h, so I added it there as well as a new row for it in hid-pidff-wrapper.c, but still no difference after rebuild and install.

I'm running a Raspberry Pi 3B

danhultg, 2024-10-01:
I found using lsusb -V that my wheel is of device class 0xFF = "Vendor Specific Class" and not HID, is that a problem?

danhultg, 2024-10-01:
Looks like I have an Xbox-compatible version of the R3 steering wheel. When plugging it in to a Windows computer it gets detected as an Xbox game controller, not a HID or a steering wheel. Perhaps this is the wrong driver for this kind of steering wheel? Not even Windows detects this wheel correctly...

Lawstorant, 2024-10-01:
You should be able to enable PC mode on this wheel. IDK how though. Still, all moza wheelbases are xbox compatible and the compatinility chip is on the wheel itself.

Sadly, we won't be able to support this xbox mode, so switching to PC mode is needed.

What about Moza Pit House? Can it detect the wheel and change it to PC mode? If yes, I could impelement PC/Xbox mode switching into Boxflat, but I would need your help to sniff the commands sent to the wheel

Lawstorant, 2024-10-01:
[image]

danhultg, 2024-10-02:
ah shit, that's the key! Switching modes worked. Thank you!!

I don't have time to support an implementation of mode switching right now, but in a couple of weeks perhaps.

danhultg, 2024-10-02:
I'm also having trouble running boxflat on my rasppi. It's nice that you have a flathub app but there's no build for armv7 so I have to build it myself, but having tons of issues with missing GTK4 components. Could take that in an issue in that repo

Lawstorant, 2024-10-02:
@danhultg yes, please. And you can close this issue as well

Lawstorant, 2024-10-17:
@JacKeTUs can be closed

closed 2024-10-17 by danhultg

warptozero, 2026-02-02:
Yes, thanks! Apparently moza support was wrong and you can indeed switch to pc mode from the mobile app with bluetooth. However you need a recent firmware for this to work and the _offline_ firmware update tool downloads are out of date on the moza site. So they will not not update to the required firmware, but using Pit House (with the _online_ installer) in a vm works.