JacKeTUs/universal-pidff #42: Arch Linux does not build under Kernel 6.12.1

LukasLichten, 2024-11-27:
Hello,  
using the AUR package the module installs fine for the lts kernel 6.6.63, but throws this error on the standard kernel:
```
==> dkms install --no-depmod universal-pidff/0.0.10.r0.gd95b73e -k 6.12.1-arch1-1

Error! Bad return status for module build on kernel: 6.12.1-arch1-1 (x86_64)
Consult /var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/make.log for more information.
==> WARNING: `dkms install --no-depmod universal-pidff/0.0.10.r0.gd95b73e -k 6.12.1-arch1-1' exited 10
```

The log file contains this:
```
DKMS make.log for universal-pidff-0.0.10.r0.gd95b73e for kernel 6.12.1-arch1-1 (x86_64)
Wed Nov 27 02:22:06 CET 2024
make -C /lib/modules/6.12.1-arch1-1/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/hid-pidff-wrapper.o
  CC [M]  /var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/hid-pidff.o
/var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/hid-pidff-wrapper.c:166:25: error: initialization of 'const __u8 * (*)(struct hid_device *, __u8 *, unsigned int *)' {aka 'const unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)'} from incompatible pointer type 'u8 * (*)(struct hid_device *, __u8 *, unsigned int *)' {aka 'unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)'} [-Wincompatible-pointer-types]
  166 |         .report_fixup = universal_pidff_report_fixup
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/hid-pidff-wrapper.c:166:25: note: (near initialization for 'universal_pidff.report_fixup')
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build/hid-pidff-wrapper.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /var/lib/dkms/universal-pidff/0.0.10.r0.gd95b73e/build] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make: *** [Makefile:5: default] Error 2
```

`dkms status` only reports the lts kernel, and that as successfully installed.  
Although, while trying to troubleshoot this, `dkms status` returned also this row:
`universal-pidff/0.0.6.r4.g9e27df1: broken` (while the lts kernel uses version `0.0.10.r0.gd95b73e`)

Reinstalling and rebooting did not resolve the issue.

markhagemann, 2024-11-27:
Yeah I am experiencing similar (CachyOS) https://github.com/JacKeTUs/universal-pidff/discussions/40

Lawstorant, 2024-11-27:
I'll include the patch in the AUR package for now

markhagemann, 2024-11-28:
```
:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                                             [----------------------------------------------------] 100%
(1/1) checking package integrity                                                           [----------------------------------------------------] 100%
(1/1) loading package files                                                                [----------------------------------------------------] 100%
(1/1) checking for file conflicts                                                          [----------------------------------------------------] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
:: Processing package changes...
(1/1) reinstalling universal-pidff-dkms-git                                                [----------------------------------------------------] 100%
:: Running post-transaction hooks...
(1/4) Arming ConditionNeedsUpdate...
(2/4) Install DKMS modules
==> dkms install --no-depmod universal-pidff/0.0.10.r0.gd95b73e -k 6.12.1-2-cachyos
==> depmod 6.12.1-2-cachyos
(3/4) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-cachyos.preset: 'default'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-cachyos -g /boot/initramfs-linux-cachyos.img
==> Using drop-in configuration file: '10-chwd.conf'
==> Starting build: '6.12.1-2-cachyos'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [microcode]
  -> Running build hook: [kms]
  -> Running build hook: [modconf]
  -> Running build hook: [block]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
loadkeys: Unable to open file: au: No such file or directory
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [plymouth]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-cachyos.img'
  -> Early uncompressed CPIO image generation successful
==> Initcpio image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux-cachyos.preset: 'fallback'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-cachyos -g /boot/initramfs-linux-cachyos-fallback.img -S autodetect
==> Using drop-in configuration file: '10-chwd.conf'
==> Starting build: '6.12.1-2-cachyos'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [microcode]
  -> Running build hook: [kms]
==> WARNING: Possibly missing firmware for module: 'ast'
  -> Running build hook: [modconf]
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'qed'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
==> WARNING: Possibly missing firmware for module: 'bfa'
==> WARNING: Possibly missing firmware for module: 'qla1280'
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'qla2xxx'
==> WARNING: Possibly missing firmware for module: 'xhci_pci_renesas'
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
loadkeys: Unable to open file: au: No such file or directory
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [plymouth]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-cachyos-fallback.img'
  -> Early uncompressed CPIO image generation successful
==> Initcpio image generation successful
(4/4) Checking which packages need to be rebuilt
```

Looks good to me. I'll test it out later tonight.

joverbeeke, 2024-11-28:
Was having the same issue, temporary used the lts kernel. Thanks for the good work

Lawstorant, 2024-11-28:
Yeah, hmm. Now that it's fixed for 6.12, it doesn't build anymore for the LTS kernel.

arabianq, 2024-12-10:
It looks like I have the same problem. I'm trying to build it on Fedora 41.

`make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: вход в каталог «/usr/src/kernels/6.11.10-300.fc41.x86_64»
  CC [M]  /opt/universal-pidff/hid-pidff-wrapper.o
/opt/universal-pidff/hid-pidff-wrapper.c:166:25: ошибка: initialization of «__u8 * (*)(struct hid_device *, __u8 *, unsigned int *)» {aka «unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)»} from incompatible pointer type «const u8 * (*)(struct hid_device *, __u8 *, unsigned int *)» {aka «const unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)»} [-Wincompatible-pointer-types]
  166 |         .report_fixup = universal_pidff_report_fixup
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/universal-pidff/hid-pidff-wrapper.c:166:25: замечание: (где-то рядом с инициализацией для «universal_pidff.report_fixup»)
make[3]: *** [scripts/Makefile.build:244: /opt/universal-pidff/hid-pidff-wrapper.o] Ошибка 1
make[2]: *** [/usr/src/kernels/6.11.10-300.fc41.x86_64/Makefile:1966: /opt/universal-pidff] Ошибка 2
make[1]: *** [Makefile:236: __sub-make] Ошибка 2
make[1]: выход из каталога «/usr/src/kernels/6.11.10-300.fc41.x86_64»
make: *** [Makefile:5: default] Ошибка 2
`

Lawstorant, 2024-12-11:
Try the previous commit d95b73e042b0cbd7053adca0324240fd0210c7da

LukasLichten, 2024-12-11:
Isn't there some sort of way to do conditional compilation, which would "change" the code dependent on the kernel version?

I am not familiar with C pragams, and not sure what dkms gives us in terms of variables/build environment info that would allow the conditional to trigger (and trigger correctly).

arabianq, 2024-12-11:
> Try the previous commit [d95b73e](https://github.com/JacKeTUs/universal-pidff/commit/d95b73e042b0cbd7053adca0324240fd0210c7da)

Thanks, it worked

Lawstorant, 2024-12-13:
> Isn't there some sort of way to do conditional compilation, which would "change" the code dependent on the kernel version?
>
> I am not familiar with C pragams, and not sure what dkms gives us in terms [...]

Actually, there is a way. I'll update the code in few minutes and create a PR

PR #45 (2024-12-13): Fix build for kernels older than 6.12
Use Kernel version macros to determine if `const` is needed for report fixups

Slight indentation fixes

Fixes #42 completely

Based on: https://github.com/berarma/new-lg4ff/commit/6100a34c182536c607af80e119d54a66c6fb2a23
closed 2024-12-13 by JacKeTUs