JacKeTUs/universal-pidff #43: Richard Burns Rally (Rally Sim Fans mod) + Moza R12 + Linux

wolfallein, 2024-12-08:
Thanks for the amazing work. I hope it gets into the mainline kernel soon.

I got my R12 FFB to work in every game I tried, except on RBR.
On Windows, it is needed to add a register entry for the FFB, as it is explained in Moza's website (https://support.mozaracing.com/en/support/solutions/articles/70000627147-richard-burns-rally-rbr-). The register entry can be download from the website.

For the R12, the content is basically:

```
[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_346E&PID_0006]
"OEMData"=hex::43,00,88,01,fe,00,00,00
"OEMName"="R12 Base" 
[HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_346E&PID_0006]
"OEMData"=hex:43,00,88,01,fe,00,00,00
"OEMName"="R12 Base"
```

I don't know what it does and why the game needs this entry for the FFB to work.
I tried to add this entry to Proton prefix via `protontricks` and it does not have any effect. Maybe I'm missing something...

I know it does not seem to be an issue with `universal-pidff` itself, since it works on other games, but I actually do not know where to ask.
Thank you!

wolfallein, 2024-12-08:
I'm closing this. The issue is with Proton:

https://github.com/ValveSoftware/Proton/issues/6702

closed 2024-12-08 by wolfallein

Lawstorant, 2025-02-20:
@wolfallein just FIY. We did extensive debugging and testing with @JacKeTUs and sadly this is an issue with the game, not Wine/proton as they work just like windows.

It's an issue with how the game initializes FFB effects. It includes all the FFB axes found on the device. The same thing happens on windows. The registry fix is on override for dinput that removes FFB flag from Y axis.

For a lot of years, Logitech, Thrustmaster and Fanatec were basically your only option and they only ever expose on FFB axis -> FFB works.

wolfallein, 2025-04-14:
Thank you very much for all your work on this issue. 

I saw that everything is fixed on the latest RSF update.
Everything is working well with my Moza R12 on latest `universal-pidff` driver and `Proton experimental`.

 It is good to not have to compile a custom `wine` anymore.

Lawstorant, 2025-04-14:
Even better. Full driver will come with Linux 6.15 but a big chunk is already queued for 6.12, 6.13 and 6.14 so you won't even have to install it manually anymore :D