JacKeTUs/universal-pidff #51: Making Releases

DaRacci, 2025-01-20:
I'm one of the maintainers for this module in [NixOS](https://github.com/NixOS/nixpkgs/blob/master/pkgs/os-specific/linux/universal-pidff/default.nix).

I've had myself subscribed to this repository for new releases but have missed them for longer than i would have liked to, due to not receiving any notifications about a new release; I was hoping that either manually or maybe via a GitHub workflow new releases could be tagged and added to GitHub so I'm able to keep this module up to date a bit better.

DaRacci, 2025-01-20:
If you want to complete this via a GitHub workflow i can assist with creating one that would check for changes in the dkms files version and create a new release once changed.

Vixea, 2025-01-20:
I'm pretty sure this driver is in the process of being upstreamed so this might not matter

JacKeTUs, 2025-01-20:
Well, upstreaming something like that is a long process... This repo isn't going anywhere. If new device will appear, it needs to be defined in headers for it to initialize and work properly, and upstreaming these changes will be tedious 

So yes, issue is relevant, I'll look into it

Vixea, 2025-01-20:
Ahh I assumes you were getting close given that it looked like some changes were made that didn't get added here

Lawstorant, 2025-04-03:
So with hid-universal-pidff now upstreamed and pushing to be included in backports for 6.14, 6.13 and 6.12 I think we could close this PR as we won't be making releases.

This repo will be of course still there for development purposes and testing new devices if they need to be added, but the suggested way of using out driver will be to just use the version in the kernel.

@DaRacci I don't know what's NixOS' policy with thins after upstreaming but I'll be adding a warning to AUR package that it's only meant as a development tool.

DaRacci, 2025-04-04:
Awesome work on getting this upstreamed, since this will be used a development repo going forward i think its best to deprecate the NixOS package, this would be done via eval errors if the kernel its compiling for already includes it, then removing the package in the 25.11 release.

The packaging process is fairly simple and if you are open to it I could make a PR to add a `flake.nix` file to this repo which then allows building on NixOS for development, I'd be happy to help maintain that inside this repo.

Lawstorant, 2025-06-20:
What do you mean?This driver was included in 6.14.3, 6.13.12, 6.12.24 as well.

JacKeTUs, 2025-06-20:
> I was just wondering if it was merged(the back ports) and what versions that's all

No problem :)
Basically now driver is fully functioning from 6.12 kernel and up. We might add some minor things / new quirks / remove quirks, or add new devices which will need quirks as well. This will definitely be included both here and in kernel later. 
Some small patches (formatting, for example) will be merged as well