JacKeTUs/universal-pidff #54: Kernel Oops due to null pointer dereference

VectorKappa, 2025-01-22:
# The problem:
Connecting a LITE STAR GT978 FF racing wheel (with same quirks set manually as all the PXNs) causes the kernel to oops. Afterwards any process using `udev-worker` hangs on `read()` calls

Weirdly enough, force feedback works for a while - `ffcfstress` moves the wheel correctly, however after quitting the process, the wheel turns sharply max to the left.

Module installed manually via `make debug; insmod hid-universal-pidff.ko`, crashes on both linux and linux-zen arch kernels.

The steering wheel itself is available as Cobra GT987 FF in Poland

# uname output:
```
Linux hephaestos 6.12.10-arch1-1.1 #1 SMP PREEMPT_DYNAMIC Sun, 19 Jan 2025 16:05:58 +0000 x86_64 GNU/Linux
```

# dmesg output:
```
Jan 22 20:00:44 hephaestos kernel: usb 1-4: new full-speed USB device number 5 using xhci_hcd
Jan 22 20:00:44 hephaestos kernel: usb 1-4: New USB device found, idVendor=11ff, idProduct=2141, bcdDevice= 0.3d
Jan 22 20:00:44 hephaestos kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 22 20:00:44 hephaestos kernel: usb 1-4: Product: GT987 FF
Jan 22 20:00:44 hephaestos kernel: usb 1-4: Manufacturer: LITE STAR
Jan 22 20:00:44 hephaestos kernel: usb 1-4: SerialNumber: 511468687777
Jan 22 20:00:44 hephaestos kernel: input: LITE STAR GT987 FF as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.0/0003:11FF:2141.0006/input/input25
Jan 22 20:00:44 hephaestos kernel: hid-generic 0003:11FF:2141.0006: unknown set_effect report layout
Jan 22 20:00:44 hephaestos kernel: hid-generic 0003:11FF:2141.0006: input,hidraw5: USB HID v1.11 Joystick [LITE STAR GT987 FF] on usb-0000:06:00.0-4/input0
Jan 22 20:00:44 hephaestos kernel: input: LITE STAR GT987 FF Keyboard as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.1/0003:11FF:2141.0007/input/input26
Jan 22 20:00:44 hephaestos kernel: input: LITE STAR GT987 FF Mouse as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.1/0003:11FF:2141.0007/input/input27
Jan 22 20:00:44 hephaestos kernel: hid-generic 0003:11FF:2141.0007: input,hidraw6: USB HID v1.10 Keyboard [LITE STAR GT987 FF] on usb-0000:06:00.0-4/input1
Jan 22 20:00:44 hephaestos kernel: input: LITE STAR GT987 FF as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.0/0003:11FF:2141.0006/input/input28
Jan 22 20:00:44 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.0006: input,hidraw5: USB HID v1.11 Joystick [LITE STAR GT987 FF] on usb-0000:06:00.0-4/input0
Jan 22 20:00:44 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.0006: device has unknown autocenter control method
Jan 22 20:00:44 hephaestos kernel: input input28: Force feedback for USB HID PID devices
Jan 22 20:00:44 hephaestos kernel: BUG: kernel NULL pointer dereference, address: 0000000000000008
Jan 22 20:00:44 hephaestos kernel: #PF: supervisor read access in kernel mode
Jan 22 20:00:44 hephaestos kernel: #PF: error_code(0x0000) - not-present page
Jan 22 20:00:44 hephaestos kernel: PGD 0 P4D 0
Jan 22 20:00:44 hephaestos kernel: Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
Jan 22 20:00:44 hephaestos kernel: CPU: 2 UID: 0 PID: 5142 Comm: (udev-worker) Tainted: G           OE      6.12.10-arch1-1.1 #1 751339c66e88745afb6185fe3bce5dadf6985c3b
Jan 22 20:00:44 hephaestos kernel: Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Jan 22 20:00:44 hephaestos kernel: Hardware name: Micro-Star International Co., Ltd MS-7C02/B450 TOMAHAWK MAX (MS-7C02), BIOS 3.C0 01/11/2022
Jan 22 20:00:44 hephaestos kernel: RIP: 0010:universal_pidff_input_configured+0x26/0x80 [hid_universal_pidff]
Jan 22 20:00:44 hephaestos kernel: Code: 90 90 90 90 f3 0f 1e fa 0f 1f 44 00 00 55 45 31 c9 45 31 c0 53 48 8b 6e 18 31 f6 bb 01 00 00 00 48 8b 85 48 01 00 00 48 89 ef <8b> 48 08 8b 50 04>
Jan 22 20:00:44 hephaestos kernel: RSP: 0018:ffffa5668661b948 EFLAGS: 00010246
Jan 22 20:00:44 hephaestos kernel: RAX: 0000000000000000 RBX: 0000000000000001 RCX: ffff98b90bdf68a8
Jan 22 20:00:44 hephaestos kernel: RDX: ffff98b839ee1868 RSI: 0000000000000000 RDI: ffff98b90bdf6800
Jan 22 20:00:44 hephaestos kernel: RBP: ffff98b90bdf6800 R08: 0000000000000000 R09: 0000000000000000
Jan 22 20:00:44 hephaestos kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff98ba08cec870
Jan 22 20:00:44 hephaestos kernel: R13: ffff98b839ee1840 R14: ffff98b839ee1840 R15: ffff98ba08cec000
Jan 22 20:00:44 hephaestos kernel: FS:  0000715fa38cb880(0000) GS:ffff98bb2ed00000(0000) knlGS:0000000000000000
Jan 22 20:00:44 hephaestos kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 22 20:00:44 hephaestos kernel: CR2: 0000000000000008 CR3: 000000036aba8000 CR4: 0000000000350ef0
Jan 22 20:00:44 hephaestos kernel: Call Trace:
Jan 22 20:00:44 hephaestos kernel:  <TASK>
Jan 22 20:00:44 hephaestos kernel:  ? __die_body.cold+0x8/0x12
Jan 22 20:00:44 hephaestos kernel:  ? page_fault_oops+0x15a/0x2e0
Jan 22 20:00:44 hephaestos kernel:  ? srso_return_thunk+0x5/0x5f
Jan 22 20:00:44 hephaestos kernel:  ? vsnprintf+0x52d/0x700
Jan 22 20:00:44 hephaestos kernel:  ? exc_page_fault+0x81/0x190
Jan 22 20:00:44 hephaestos kernel:  ? asm_exc_page_fault+0x26/0x30
Jan 22 20:00:44 hephaestos kernel:  ? universal_pidff_input_configured+0x26/0x80 [hid_universal_pidff 55b49ae4cd884535a75fc35c5a6e3d5760c319cb]
Jan 22 20:00:44 hephaestos kernel:  hidinput_connect+0x54b/0xd90
Jan 22 20:00:44 hephaestos kernel:  ? __kmalloc_noprof+0x189/0x400
Jan 22 20:00:44 hephaestos kernel:  hid_connect+0x3c3/0x610
Jan 22 20:00:44 hephaestos kernel:  ? srso_return_thunk+0x5/0x5f
Jan 22 20:00:44 hephaestos kernel:  hid_hw_start+0x40/0x70
Jan 22 20:00:44 hephaestos kernel:  universal_pidff_probe+0x2f/0x60 [hid_universal_pidff 55b49ae4cd884535a75fc35c5a6e3d5760c319cb]
Jan 22 20:00:44 hephaestos kernel:  ? hid_lookup_quirk+0x7b/0x1e0
Jan 22 20:00:44 hephaestos kernel:  hid_device_probe+0x111/0x1e0
Jan 22 20:00:44 hephaestos kernel:  really_probe+0xde/0x3b0
Jan 22 20:00:44 hephaestos kernel:  ? pm_runtime_barrier+0x54/0x90
[... 25 lines cut ...]
Jan 22 20:00:44 hephaestos kernel: Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff>
Jan 22 20:00:44 hephaestos kernel: RSP: 002b:00007ffe70f2c2f8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
Jan 22 20:00:44 hephaestos kernel: RAX: ffffffffffffffda RBX: 00005e7a7d100a10 RCX: 0000715fa32621fd
Jan 22 20:00:44 hephaestos kernel: RDX: 0000000000000004 RSI: 0000715fa29c94ac RDI: 0000000000000012
Jan 22 20:00:44 hephaestos kernel: RBP: 00007ffe70f2c3a0 R08: 00005e7a7d0dee60 R09: 00005e7a7cf04cb0
Jan 22 20:00:44 hephaestos kernel: R10: 00005e7a7d0dee90 R11: 0000000000000246 R12: 0000715fa29c94ac
Jan 22 20:00:44 hephaestos kernel: R13: 0000000000020000 R14: 0000000000000000 R15: 00005e7a7d0cd730
Jan 22 20:00:44 hephaestos kernel:  </TASK>
Jan 22 20:00:44 hephaestos kernel: Modules linked in: hid_universal_pidff(OE+) rfcomm snd_seq_dummy snd_seq_midi snd_hrtimer snd_seq_midi_event snd_seq uinput rpcrdma rdma_cm iw_cm ib_cm>
Jan 22 20:00:44 hephaestos kernel:  videobuf2_common aesni_intel snd_pcm realtek sp5100_tco gf128mul videodev crypto_simd mdio_devres snd_timer cryptd rfkill i2c_piix4 crc16 wmi_bmof rap>
Jan 22 20:00:44 hephaestos kernel: CR2: 0000000000000008
Jan 22 20:00:44 hephaestos kernel: ---[ end trace 0000000000000000 ]---
Jan 22 20:00:44 hephaestos kernel: RIP: 0010:universal_pidff_input_configured+0x26/0x80 [hid_universal_pidff]
Jan 22 20:00:44 hephaestos kernel: Code: 90 90 90 90 f3 0f 1e fa 0f 1f 44 00 00 55 45 31 c9 45 31 c0 53 48 8b 6e 18 31 f6 bb 01 00 00 00 48 8b 85 48 01 00 00 48 89 ef <8b> 48 08 8b 50 04>
Jan 22 20:00:44 hephaestos kernel: RSP: 0018:ffffa5668661b948 EFLAGS: 00010246
Jan 22 20:00:44 hephaestos kernel: RAX: 0000000000000000 RBX: 0000000000000001 RCX: ffff98b90bdf68a8
Jan 22 20:00:44 hephaestos kernel: RDX: ffff98b839ee1868 RSI: 0000000000000000 RDI: ffff98b90bdf6800
Jan 22 20:00:44 hephaestos kernel: RBP: ffff98b90bdf6800 R08: 0000000000000000 R09: 0000000000000000
Jan 22 20:00:44 hephaestos kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff98ba08cec870
Jan 22 20:00:44 hephaestos kernel: R13: ffff98b839ee1840 R14: ffff98b839ee1840 R15: ffff98ba08cec000
Jan 22 20:00:44 hephaestos kernel: FS:  0000715fa38cb880(0000) GS:ffff98bb2ed00000(0000) knlGS:0000000000000000
Jan 22 20:00:44 hephaestos kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 22 20:00:44 hephaestos kernel: CR2: 0000000000000008 CR3: 000000036aba8000 CR4: 0000000000350ef0
Jan 22 20:00:44 hephaestos kernel: note: (udev-worker)[5142] exited with irqs disabled
```

JacKeTUs, 2025-01-23:
Can you, please, check branch `54-kernel-oops-due-to-null-pointer-dereference`? I added some checks to the input configuration routine.

VectorKappa, 2025-01-23:
> Hello! Can you, please, dump HID descriptor of the device and send it here?

Sure thing! Here's the wireshark (abbreviated) output:

```
HID Report
    Usage Page (Generic Desktop Controls)
        Header
        Usage Page: Generic Desktop Controls (0x01)
    Usage (Joystick)
        Header
        Usage: Joystick (0x04)
    Collection (Application)
        Header
        Collection type: Application (0x01)
        Report ID (0x01)
        Usage (Pointer)
        Collection (Physical)
        Usage (Z)
        Usage (Rx)
        Usage (Ry)
        Usage (Rz)
        Usage (Slider)
        Logical Minimum (0)
        Logical Maximum (65535)
        Physical Maximum (65535)
        Report Size (16)
        Report Count (5)
        Input (Data,Var,Abs)
        Usage (Hat switch)
        Logical Minimum (1)
        Logical Maximum (8)
        Physical Minimum (0)
        Physical Maximum (315)
        Unit (0x14)
        Report Size (8)
        Report Count (1)
        Input (Data,Var,Abs)
        Physical Maximum (0)
        Unit (0x00)
        Usage Page (Button)
        Usage Minimum (0x01)
        Usage Maximum (0x18)
        Logical Minimum (0)
        Logical Maximum (1)
        Physical Maximum (1)
        Report Size (1)
        Report Count (24)
        Input (Data,Var,Abs)
        Report Count (16)
        Input (Const,Var,Abs)
        Usage Page (Physical Interface Device (PID))
        Usage (PID State Report)
        Collection (Logical)
        Usage (Set Effect Report)
        Collection (Logical)
        Usage (Set Envelope Report)
        Collection (Logical)
        Usage (Set Condition Report)
        Collection (Logical)
        Usage (Set Periodic Report)
        Collection (Logical)
        Usage (Set Constant Force Report)
        Collection (Logical)
        Usage (Set Ramp Force Report)
        Collection (Logical)
        Usage (Custom Force Data Report)
        Collection (Logical)
        Usage (Download Force Sample)
        Collection (Logical)
        Usage Page (Physical Interface Device (PID))
        Usage (Effect Operation Report)
        Collection (Logical)
        Usage (PID Block Free Report)
        Collection (Logical)
        Usage (PID Device Control)
        Collection (Logical)
        Usage (Device Gain Report)
        Collection (Logical)
        Usage (Set Custom Force Report)
        Collection (Logical)
        Usage (Create New Effect Report)
        Collection (Logical)
        Usage Page (Physical Interface Device (PID))
        Usage (PID Block Load Report)
        Collection (Logical)
        Usage (PID Pool Report)
        Collection (Logical)
        End Collection

```
I can also provide the full GET_DESCRIPTOR Response packet

Lawstorant, 2025-01-23:
Might be related https://lkml.iu.edu/hypermail/linux/kernel/2411.0/04043.html?

Lawstorant, 2025-01-23:
I think just relying on the pidff error is okay but we don't have to fail. We control what devices bind to universal-pidff so we can just allow pidff_init failure

Lawstorant, 2025-01-23:
I proposed some changes around that in the latest commit to #55

Lawstorant, 2025-01-23:
@VectorKappa are you sure this is the only descriptor that's detected when plugging in this device?

JacKeTUs, 2025-01-23:
Hmm, but with new branch we're getting now this
```
Jan 23 18:41:06 hephaestos kernel: input input28: Force feedback for USB HID PID devices
Jan 23 18:41:06 hephaestos kernel: input input28: Active pidff quirks: 0x12
```
So ffb should be ok

JacKeTUs, 2025-01-23:
I think this device propagates itself as joystick and keyboard
And previous crashes was related to X axis and no absinfo structure (from second device)

Lawstorant, 2025-01-23:
Hmm, let me add a check for joystick in the input mapping function

VectorKappa, 2025-01-23:
Even better, it's Joystick, Mouse AND a keyboard!

VectorKappa, 2025-01-23:
I should've mentioned this earlier - i have PIDFF_QUIRK_NO_DELAY_EFFECT, and PIDFF_QUIRK_PERIODIC_SINE_ONLY quirks enabled

Lawstorant, 2025-01-23:
>(with same quirks set manually as all the PXNs)
Yeah, you did. No worries.

Hmm, I guess we could check the usage type during probe. After hid_hw_start we should be able to check `hid_device->collection[]->usage` for `HID_GD_JOYSTICK`

VectorKappa, 2025-01-23:
PR #55 Works!
However, the supplied effect-test file only works for constant forces, and (presumably) for friction and inertia. Periodics dont work

Lawstorant, 2025-01-23:
They do not work at all or do they just not rumble and turn for example? You could try it without the PERIODIC_SINE_ONLY quirk

VectorKappa, 2025-01-23:
On commit [75d7effa49a884a357c13486f9a8a112bffc1f09](https://github.com/JacKeTUs/universal-pidff/commit/75d7effa49a884a357c13486f9a8a112bffc1f09), Assetto Corsa works! Both rumble and forces when losing control at high speeds. Currently waiting for BeamNG to bake its shaders, will keep you posted

Lawstorant, 2025-01-23:
That's probably because most games do rumble themselves with constant force. Most games use just the constant force and maybe damper/friction.

VectorKappa, 2025-01-23:
Here's the dmesg from your latest commit:
```
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 0: usage 288
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 1: usage 289
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 2: usage 290
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 3: usage 291
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 4: usage 292
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 5: usage 293
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 6: usage 294
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 7: usage 295
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 8: usage 296
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 9: usage 297
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 10: usage 298
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 11: usage 299
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 12: usage 300
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 13: usage 301
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 14: usage 302
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 15: usage 303
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 16: usage 624
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 17: usage 625
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 18: usage 626
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 19: usage 627
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 20: usage 628
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 21: usage 629
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 22: usage 630
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Button 23: usage 631
Jan 23 20:36:09 hephaestos kernel: input: LITE STAR GT987 FF as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.0/0003:11FF:2141.000A/input/input48
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: input,hidraw5: USB HID v1.11 Joystick [LITE STAR GT987 FF] on usb-0000:06:00.0-4/input0
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Joystick usage found in the descriptor
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: starting pid init
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x21 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x5a from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x5f from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x6e from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x73 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x74 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x77 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x90 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x96 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x7d from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0xab from collection array
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x89 from field->logical
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: found usage 0x7f from field->logical
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 0 at 0->0
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 1 at 2->0
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 2 at 3->0
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 3 at 4->0
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 4 at 2->1
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: found 5 at 6->0
Jan 23 20:36:09 hephaestos kernel: hid_universal_pidff: maxusage and report_count do not match, skipping
[... cut ...]
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: Universal pidff driver loaded sucesfully!
Jan 23 20:36:09 hephaestos kernel: input: LITE STAR GT987 FF as /devices/pci0000:00/0000:00:01.3/0000:06:00.0/usb1/1-4/1-4:1.1/0003:11FF:2141.000B/input/input49
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000B: input,hidraw7: USB HID v1.10 Keyboard [LITE STAR GT987 FF] on usb-0000:06:00.0-4/input1
Jan 23 20:36:09 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000B: Joystick usage not found in the descriptor
```

VectorKappa, 2025-01-23:
Rundown of the effect-test:

# Constant force left - Moves left
# Constant force right - Moves right
# Spring - Nothing
# Damper - Nothing
# Friction - Nothing
# Inertia - Slightly less resistance than the previous 3

------- nothing from here
# Periodic sine
# Periodic square
# Periodic triangle
# Periodic saw up
# Periodic saw down
# Sine constant force emulation left
# Sine constant force emulation right
# Back to center
# Test end

Dmesg during ffbplay:
```
Jan 23 20:40:40 hephaestos kernel: hid_universal_pidff: calculated from 65535 to 255
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: resetting device
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: enable actuators
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: create_new_effect sent, type: 1
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: pid_block_load requested
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: device reported free memory: 1200 bytes
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 1
Jan 23 20:40:40 hephaestos kernel: hid_universal_pidff: calculated from 8000 to 2441
Jan 23 20:40:40 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: uploaded
Jan 23 20:40:43 hephaestos kernel: hid_universal_pidff: calculated from -8000 to -2441
Jan 23 20:40:43 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: uploaded
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: starting to erase 0/1
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 0
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: resetting device
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: enable actuators
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: create_new_effect sent, type: 8
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: pid_block_load requested
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: device reported free memory: 1200 bytes
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 1
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 0 to 0
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 2000 to 610
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 2000 to 610
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 65535 to 10000
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 65535 to 10000
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 0 to 0
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 0 to 0
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 4832 to 1474
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from -13635 to -4161
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 29037 to 4430
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 0 to 0
Jan 23 20:40:46 hephaestos kernel: hid_universal_pidff: calculated from 29037 to 4430
Jan 23 20:40:46 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: uploaded
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: starting to erase 0/1
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 0
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: resetting device
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: enable actuators
Jan 23 20:40:49 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: DEVICE_CONTROL is an array
[... 224 lines cut ...]
Jan 23 20:41:20 hephaestos kernel: hid_universal_pidff: calculated from 0 to 0
Jan 23 20:41:20 hephaestos kernel: hid_universal_pidff: calculated from 29037 to 4430
Jan 23 20:41:20 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: uploaded
Jan 23 20:41:20 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: create_new_effect sent, type: 8
[... cut ...]
Jan 23 20:41:20 hephaestos kernel: hid_universal_pidff: calculated from 29037 to 4430
Jan 23 20:41:20 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: uploaded
Jan 23 20:41:25 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: starting to erase 1/2
Jan 23 20:41:25 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 1
Jan 23 20:41:25 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: starting to erase 0/1
Jan 23 20:41:25 hephaestos kernel: hid-universal-pidff 0003:11FF:2141.000A: current effect count: 0
```

Lawstorant, 2025-01-23:
I guess they defined the effects but they just don't work. It's a cheap gear-based wheel after all. Maybe windows is strict about having all the effects as well, that's why they're there, who knows.

JacKeTUs, 2025-01-23:
Well, if you got "Slightly less resistance on inertia" - that does mean damper and friction are working.

Majority of the games using ConstantForce + Damper (Assetto Corsa titles, for example), so you should be fine playing it

Could you try the wheel on Windows and see if all effects are working from it? You can use WheelCheck.exe from [here](https://github.com/JacKeTUs/ffb-experiments)

VectorKappa, 2025-01-23:
Well, i think i'll mark the original issue as solved in #55 :)
I can't thank you people enough!

I ran ForceTest.exe from fs-force.com.
All the effects except Damping and Grooves worked there.
I can provide further testing if needed.

BeamNG.Drive and Assetto Corsa work out of the box, DiRT 3 needs some setting tweaking.

Lawstorant, 2025-01-24:
Just one more question. Which axis is the steering on your wheel?

VectorKappa, 2025-01-28:
It's the X axis according to BeamNG, Axis 1 according to antimicroX.

Lawstorant, 2025-01-28:
Ok thanks, so it must've been missing from the other "devices" then. No need to adjust anything when it comes to fuzz/flat values.

PR #55 (2025-02-01): Code sync with upstreaming effort
Compared to main:
Period rescaling
Updated clamping

`DEVICE_CONTROL` variable/array detection
Fixed reset, actuators modifications
Reset on effect uploads to an "empty" device, not just on init

Simplified `effect_upload` function
Moved some magic values to defines
Quirks renamed and repositioned
Clearer envelope check

Hid ids indented with tabs
Comment styling changes

Added support for LITE STAR GT978 FF
Closes #54 
Closes #56 

Test file changes:
Decrease magnitude of PERIODIC effects
Longer return spring
closed 2025-02-01 by JacKeTUs