JacKeTUs/universal-pidff #56: Null-pointer deref if optional HID reports are not present

Lawstorant, 2025-01-23:
From LKML:
https://lore.kernel.org/r/lkml/CAL-gK7f5=R0nrrQdPtaZZr1fd-cdAMbDMuZ_NLA8vM0SX+nGSw@mail.gmail.com/#t

When initializing a HID PID device, hid-pidff.c checks for eight required
HID reports and five optional reports. If the eight required reports are
present, the hid_pidff_init() function then attempts to find the necessary
fields in each required or optional report, using the pidff_find_fields()
function. However, if any of the five optional reports is not present,
pidff_find_fields() will trigger a null-pointer dereference.

I recently implemented the descriptors for a USB HID device with PID
force-feedback capability. After implementing the required report
descriptors but not the optional ones, I got an OOPS from the
pidff_find_fields function. I saved the OOPS from my Ubuntu installation,
and have attached it here. I later reproduced the issue on 6.11.6.

I was able to work around the issue by having my device present all of the
optional report descriptors as well as all of the required ones.

Reported-by: Nolan Nicholson <nolananicholson@gmail.com>

[trace.txt](https://github.com/user-attachments/files/18526355/trace.txt)

Lawstorant, 2025-01-24:
I contacted Nolan via e-mail to hopefully get some more info about this

Lawstorant, 2025-01-26:
This is what this bug report is about:
```c
/* Report usage table used to put reports into an array */

#define PID_SET_EFFECT		0
#define PID_EFFECT_OPERATION	1
#define PID_DEVICE_GAIN		2
#define PID_POOL		3
#define PID_BLOCK_LOAD		4
#define PID_BLOCK_FREE		5
#define PID_DEVICE_CONTROL	6
#define PID_CREATE_NEW_EFFECT	7

#define PID_REQUIRED_REPORTS	7

#define PID_SET_ENVELOPE	8
#define PID_SET_CONDITION	9
#define PID_SET_PERIODIC	10
#define PID_SET_CONSTANT	11
#define PID_SET_RAMP		12
static const u8 pidff_reports[] = {
	0x21, 0x77, 0x7d, 0x7f, 0x89, 0x90, 0x96, 0xab,
	0x5a, 0x5f, 0x6e, 0x73, 0x74
};
```

Reports after `PID_REQUIRED_REPORTS` are optional. Same goes for custom force reports. I'll check out what happens when some of them are missing as devices are required to support all effect types.

Lawstorant, 2025-01-27:
I replaced `SET_RAMP` identifier `0x74 -> 0x4f`. This triggered a null pointer reference. It seems that `pidff_find_fields` never checks if the passed report isn't NULL. If a device doesn't implement even one of these reports, it will trigger this error.

After adding a simple check at the start, the error is gone. I'd love Nolan Nicholson to test this, but  I didn't receive any reply.

NolanNicholson, 2025-01-28:
Hi!

1) Thank you for taking up this mantle, and for your patience with my reply.
2) You are exactly correct - those fields after PID_REQUIRED_REPORTS are the ones that will trigger the null-pointer deref if you try to include them.

I'd be happy to test the fix if needed.

Lawstorant, 2025-01-28:
Thanks for taking the time to get here! 

Hmm yes, it's a fairly simple fix. You can just `make debug` this repository and load the module or I could attach a kernel patch for you. Whatever is easier.

PR #55 (2025-02-01): Code sync with upstreaming effort
Compared to main:
Period rescaling
Updated clamping

`DEVICE_CONTROL` variable/array detection
Fixed reset, actuators modifications
Reset on effect uploads to an "empty" device, not just on init

Simplified `effect_upload` function
Moved some magic values to defines
Quirks renamed and repositioned
Clearer envelope check

Hid ids indented with tabs
Comment styling changes

Added support for LITE STAR GT978 FF
Closes #54 
Closes #56 

Test file changes:
Decrease magnitude of PERIODIC effects
Longer return spring
closed 2025-02-01 by JacKeTUs