JacKeTUs/universal-pidff #64: Steam deck support

DarkShtir, 2025-02-02:
Hello)

I have one question, for someone who is more competent in Linux. 

Maybe someone installed this driver on steam os, on a steam deck? 
Because steam os is a more locked system than other distributors, and the install section from readme does not fit. 

Best regards, and thank you for your work!)

linkworld, 2025-02-02:
I would also be very interested to have any kind of directions on how to install these drivers on Steam OS.

JacKeTUs, 2025-02-03:
I thought Deck could install dkms modules when filesystem is unlocked, so it should be easy task. May be we can have couple of "helper" scripts to assist Deck users with installation

linkworld, 2025-02-03:
I am sorry, I meant you of course Tomasz - @Lawstorant. I have a Moza R3 ordered yesterday so it would be great to check how it works with the Deck - especially if someone much more experienced than myself would try it out and give some hints :-) 

@JacKeTUs from what I was reading - DKMS would be the best way to go so that the driver won't be deleted after the Deck update. But I am not sure if I know how to install DKMS and the module afterwards. That's why @Lawstorant 's help and yours would be much appreciated.

DarkShtir, 2025-02-03:
'...Deck could install dkms modules when filesystem is unlocked'
Yes but no)

I tried to unlock the file system, and found some instructions about installing DKMS, but the last topic about this was about installing the Xbox app on deck since 2022-2023. And this topic has point about closing install DKMS opportunity.

But I'm a novice in Linux, and especially steam os, and maybe don't understand this topic correctly.

And I can help with testing the solution on deck)

Best regards)

Lawstorant, 2025-02-03:
I'll create a short tutorial + a script that will do everything automagically. Since I maintin the AUR package, the script can just download it's PKGBUILD to install it. This way we'll get all the necessary dependencies apart from headers, but these can be detected based on the kernel name

linkworld, 2025-02-03:
Wow - that's amazing. I am looking forward to your tutorial. BTW it would probably the first on the web - regarding Moza+ the Deck.

Lawstorant, 2025-02-03:
Let's hope it won't be needed in a few months anyway

linkworld, 2025-02-03:
Can you explain a little more? Why it won't be needed? Is the driver going to be incorporated into Steam Deck ?

Lawstorant, 2025-02-03:
The driver should be included in the Linux 6.15. If Valve upgrades to that version this year, it will contain it. Or valve could backport it, but I doubt they care about steering wheels

linkworld, 2025-02-03:
Ok. Got it. For the time being your tutorial would be a life saver. Where are you planning to publish it? Here on GitHub or somewhere else?

Lawstorant, 2025-02-03:
Instructions (manual and automatic) will be added to the README here in #65 

Could you try out my install script?
```bash
# first, set password
passwd deck

# run script
sudo sh -c "$(curl -fsSL https://raw.githubusercontent.com/Lawstorant/hid-universal-pidff/steam-deck/docs/steam-deck-install.sh)"
```

linkworld, 2025-02-03:
I am expecting to have my Moza R3 tomorrow, so either tomorrow or the day after I will surely give it a try. Thanks so much.

DarkShtir, 2025-02-04:
All works!)

Thank you so much!)

I'm very appreciative of your work and help)

Lawstorant, 2025-02-06:
@linkworld how's your R3?

closed 2025-02-06 by JacKeTUs

Lawstorant, 2025-02-06:
I'll be honest, I'm not quite sure what will happen, but I guessing at the least, you'll have to reinstall module through dkms, at the worst run the script again. I may do some testing on where to best put scripts to survive system updates so it would be easier to reinstall. Maybe, it will be as easy as just running `sudo reinstall-universal-pidff`.

I guess I'll hear from you when that happens :) Let's just hope Valve updates to a kernel that contains upstream patches in the future and there will be no need to install anything.

PR #65 (2025-02-06): Add Steam Deck installation instructions and script
Closes #64 

Let's see if it works besides my device