JacKeTUs/universal-pidff #74: SRP-Lite pedals sometimes remaining at 1% pressure

rpop0, 2025-03-11:
When playing BeamNG, I noticed that my throttle and brake pedals are sometimes remaining at 1% brake pressure. Now generally this is not an issue in most games but in BeamNG, if there is any brake pressure applied, automatic cars don't switch gears.

I am not completely sure that this is the best place to make this issue, but I tried the same exact setup on Windows and the issue was not present, so I figured it's something with the linux driver. Let me know if more info is needed.

JacKeTUs, 2025-03-14:
I'm assuming your pedals are connected through Moza wheelbase?
I think it's the issue with calibrating the pedals, and i think it's best to create issue in boxflat repo, which is managing Moza devices, including calibration. 
@Lawstorant, fyi

Lawstorant, 2025-03-14:
@rpop0 This driver is only for the wheelbases. Please create an issue [here](https://github.com/Lawstorant/boxflat) and I'll help you debug.

@JacKeTUs you can close this I think as this is unrelated to `hid-universal-pidff`

closed 2025-03-14 by JacKeTUs