JacKeTUs/universal-pidff #76: Unable to get games to see VRS wheel in games (Arch)

lithiumfox, 2025-03-29:
I figured I'd make this a formal issue since I'll have some free time this week.

I have tried this in a few distros: Manjaro, Bazzite (Fedora), CachyOS (Arch) and then spun up my own Arch from archinstall script and have been unable to get my VRS wheel to be detected in any game.

I am currently using that Arch build using an ISO from earlier this month.

My current device setup:

-  VRS DPF20 Wheelbase
-  Fanatec ClubSport V3 Pedals
-  Thustmaster Sparco Wheel QR Adapter -> Keyboard/Generic Joystick
-  Thrustmaster TH8A
-  Thrustmaster Sparco Handbrake

After testing, I can get 4 of 5 devices detected via multiple methods of install. I have tried the following methods of installation:

- Manual DKMS Install
- Arch Pacman install (AUR install)
- Modified SteamDeck / Manual Install (since commands were similar, tried to see if anything would get it to work) [Making sure headers were there and such]
- Tried using `simracing-essentials` AUR https://aur.archlinux.org/packages/simracing-essentials)

I've set `11-vrs.rules` and tried a few different combinations of the SDL/Hidraw setups in Proton Experimental (The only way I can really get iRacing to load. There may be another proton that works but I'm unsure.)

So far, I can see all Thrustmaster devices in the Linux `KDE game controller` window, and Fanatec pedals appears in the `Wine Control Panel -> Game Controllers panel` when setting the hidraw settings. 

If there's anything else I can provide, let me know.

Lawstorant, 2025-03-31:
Could you show us the output from evtest? Or maybe a screenshot from https://hardwaretester.com/gamepad?

Moreover, did you try the included `effect-test.ffb`?

lithiumfox, 2025-04-01:
FFB was working.

I tried adding the udev rule first to test that as it's generally a bit less work.

Was detected after adding `--buttons 1` with [protopedal](https://gitlab.com/openirseny/protopedal)

Obviously requires that it's continuously running in the background, but that's a bit... annoying? annoying is probably the way to put it. Silly that they don't count devices without a button as a joystick.

Lawstorant, 2025-04-01:
Well, in USB HID terms, Axes are just General Desktop Usage. Joystick Usage Page only defines buttons soo without buttons, it's kind of hard to decide if a device is a joystick or not.

lithiumfox, 2025-04-12:
Oh, if I could get those udev rules that would be great, or if they could be included as part of simracing-essentials. (Also were drivers included in 6.14 kernel?)

JacKeTUs, 2025-04-15:
As far as i understand things, driver should be included in earlier versions of the driver soon, probably next release. It's already in 'stable' git.

Also, game does not recognize wheelbase even with PROTON_ENABLE_HIDRAW variable?

Lawstorant, 2025-05-22:
Yeah, this udev rule should just work:
```
SUBSYSTEM=="input", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a355", ENV{ID_INPUT_JOYSTICK}="1"
```

JacKeTUs, 2025-05-23:
@lithiumfox, we found a way to reliably force joystick type to the device

Create new hwdb entry by creating file `/etc/udev/hwdb.d/99-vrs.hwdb` with this content:
```
# VRS DFP
id-input:modalias:input:*v0483pa355*
 ID_INPUT_ACCELEROMETER=0
 ID_INPUT_JOYSTICK=1
```

Update hwdb database with `sudo systemd-hwdb update` and reconnect the device / retrigger with `sudo udevadm trigger`

This will set joystick type for udev and later udev with their rules will create symlinks which are responsible for joystick detection in various software, including SDL and Wine.
You will no longer need PROTON_ENABLE_HIDRAW variable for the games.
We already had some folks with VRS wheelbases who tested it and confirmed that it worked for them.

JacKeTUs, 2025-06-30:
I'm now can test it myself, and i can confirm that hwdb rules will help users - no need to set any other mode (SDL devices, hidraw....)

[image]

There is AC Evo with nothing but proton debugging stuff, and hwdb rule, recognizing DFP15 without an issue