JacKeTUs/universal-pidff #77: Support for LiteStar v995 (Ardor Silverstone v2)

legin449, 2025-04-20:
Hi, i have a wheel Ardor Silverstone v2 which is basically LiteStar v995 with Ardor logo
I have managed to get the wheel's ffb to work just renaming PXN V10's pid to pid of my wheel (0x3245 -> 0xf045)
In some games wheel is working almost perfectly (ETS, ATS, BeamNG) but i don't have vibration for some reason
But i still have issues in other games, for example in WRC9 wheel is not detecting at all, in Forza Horizon 5 autocentering is not working

In dmesg i have these logs:

```
[  +0.085459] input: LiteStar ARDOR GAMING Silverstone V2 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-9/1-9.4/1-9.4:1.0/0003:11FF:F045.0011/input/input33
[  +0.000100] hid-universal-pidff 0003:11FF:F045.0011: input,hidraw8: USB HID v1.11 Joystick [LiteStar ARDOR GAMING Silverstone V2] on usb-0000:02:00.0-9.4/input0
[  +0.021428] hid-universal-pidff 0003:11FF:F045.0011: device has unknown autocenter control method
[  +0.000006] input input33: Force feedback for USB HID PID devices by Anssi Hannula <anssi.hannula@gmail.com>
[  +0.000002] hid-universal-pidff 0003:11FF:F045.0011: Universal pidff driver loaded successfully!
[  +0.018138] input: LiteStar ARDOR GAMING Silverstone V2 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-9/1-9.4/1-9.4:1.1/0003:11FF:F045.0012/input/input34
[  +0.096522] hid-universal-pidff 0003:11FF:F045.0012: input,hidraw9: USB HID v1.10 Keyboard [LiteStar ARDOR GAMING Silverstone V2] on usb-0000:02:00.0-9.4/input1
```

legin449, 2025-04-21:
Hi
I found out that PXN V10's firmware is suitable for my device and with universal-pidff vibration and autocentering is working
Also i don't have message about "autocentering control method" in dmesg
Now i have an issue that vibration is too strong in comparison with original firmware and windows, also buttons are located differently and one of the button is always pressed (checked it in jstest)
So i think that V10 and my device have the same base but slightly different methods

Lawstorant, 2025-04-22:
Hmm, just adding your wheel to the driver should be enough. What distro are you on?

legin449, 2025-04-22:
Its archlinux, driver was build with dkms from this repo

Lawstorant, 2025-04-22:
Apart from that, it would be great if you could build manually with debug.
```shell
make debug
sudo make load
```

Lawstorant, 2025-04-22:
Hmm, doesn't seem like it works. I'll add a branch for your wheel after work and we'll see

legin449, 2025-04-24:
Thank you
Let me know if you'll need any help with testing

Lawstorant, 2025-04-24:
Available in this PR: #78

legin449, 2025-04-24:
Tried new driver, ran effect-tests
On "periodic" steps i can hear something inside the wheel but nothing really happening
"sine" steps looks like not working
Also "back to center" effect too weak
Maybe i misunderstand how these effects shoud work so i recorded a video:
[gdrive video link here](https://drive.google.com/file/d/1o8j4IqYq3G_CC-s7f-_VJyC4kkTW-q3F/view?usp=sharing)

Logs with new driver and logs after ffbplay attached below:

```
Apr 24 20:43:07 archlinux kernel: usb 1-9.4: USB disconnect, device number 11
Apr 24 20:43:09 archlinux kernel: usb 1-9.4: new full-speed USB device number 12 using xhci_hcd
Apr 24 20:43:10 archlinux kernel: usb 1-9.4: New USB device found, idVendor=11ff, idProduct=f045, bcdDevice= 0.32
Apr 24 20:43:10 archlinux kernel: usb 1-9.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 24 20:43:10 archlinux kernel: usb 1-9.4: Product: ARDOR GAMING Silverstone V2
Apr 24 20:43:10 archlinux kernel: usb 1-9.4: Manufacturer: LiteStar
Apr 24 20:43:10 archlinux kernel: usb 1-9.4: SerialNumber: 519678917777
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 0: usage 288
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 1: usage 289
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 2: usage 290
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 3: usage 291
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 4: usage 292
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 5: usage 293
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 6: usage 294
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 7: usage 295
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 8: usage 296
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 9: usage 297
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 10: usage 298
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 11: usage 299
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 12: usage 300
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 13: usage 301
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 14: usage 302
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 15: usage 303
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 16: usage 624
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 17: usage 625
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 18: usage 626
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 19: usage 627
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 20: usage 628
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 21: usage 629
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 22: usage 630
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Button 23: usage 631
Apr 24 20:43:10 archlinux kernel: input: LiteStar ARDOR GAMING Silverstone V2 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-9/1-9.4/1-9.4:1.0/0003:11FF:F045.000D/input/input30
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: input,hidraw8: USB HID v1.11 Joystick [LiteStar ARDOR GAMING Silverstone V2] on usb-0000:02:00.0-9.4/input0
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: PID usage page found
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: starting pid init
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x21 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x5a from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x5f from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x6e from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x73 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x74 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x77 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x90 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x96 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x7d from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0xab from collection array
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x89 from field->logical
Apr 24 20:43:10 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: found usage 0x7f from field->logical
Apr 24 20:43:10 archlinux kernel: hid_universal_pidff: found 0 at 0->0
[... cut ...]
Apr 24 20:44:38 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Requesting playback on FF effect 0
Apr 24 20:44:38 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Playing PID effect 1
Apr 24 20:44:38 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: starting to erase 0/1
Apr 24 20:44:38 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: Playing PID effect 1
Apr 24 20:44:38 archlinux kernel: hid-universal-pidff 0003:11FF:F045.000D: current effect count: 0
```

Lawstorant, 2025-04-24:
Do you have any windows application to control the wheel? We had issues with another PXN rebrand and it turns out the default settings had many ffb effects gain set to 0 and that's why they didn't work.

apart from that, let me remove the sine quirk and we'll see. Seems like the wheel doesn't really like periodic effects that much.

legin449, 2025-04-24:
i have an application which works with the wheel via bluetooth
i can try to max all the settings out and rerun test

Lawstorant, 2025-04-24:
I updated the branch with some changes now.

legin449, 2025-04-24:
It's an android app:
[image]

legin449, 2025-04-24:
Looks like periodic effect working are working now
Spring effect and Sine are missing
'Back to center' is sending the wheel to center position, but in game its not fully working for some reason (for example, if I turn the steering wheel all the way to the left and press the gas pedal, the steering wheel turns towards the center just a little bit and stops)

Lawstorant, 2025-04-24:
1. Spring and back to center are the same :P It's a spring effect. I have no idea why it doesn't work the first time though.
2. Does the "periodic sine" effect work? Is it just the sine emulation at the end that's broken?
3. Yeah, games do what they need. Your wheel won't always just return straight to center.

legin449, 2025-04-24:
1. I just found out that spring effect works strange, if i don't touch the wheel its not centering, but if i turn in a bit left after 'Constant force right' it's centering
2. Yep, emulation isn't working
3. It worked on windows differently, correctly i would say (also with pxn v10 firmware on linux it worked correctly)

Lawstorant, 2025-04-24:
Constant force works so IDK, maybe lower the damper/friction. Apart from that, I'd say most things works as they should. Again, we don't really do anything to the FFB itself.

With V10 firmware, I don't know why vibration would be gone, but probably some mismatch as we won't be able to fix that.

legin449, 2025-04-24:
I can check it on windows
How can i make sure that emulation is working?

Lawstorant, 2025-04-24:
Okay, emulation is kind of a brain shortcut there but basically, if you have force feedback in these games (the main one, not vibrations etc) it means it works.

legin449, 2025-04-24:
In race 07 have ffb even on linux

Lawstorant, 2025-04-24:
Okay, so something must be off with the effect-test script.

So, it seems like there's actually nothing to do to support your wheel? Just works with the generic driver now.

legin449, 2025-04-24:
if we can't fix forza horizon 5 wheel centering so topic can be closed
anyway i can still use pxn's firmware for that

Lawstorant, 2025-04-24:
FH5 has weird FFB. I never could make it work with my Moza R9, even on windows :P

closed 2025-04-24 by legin449