JacKeTUs/universal-pidff #79: Moza R3 Weird Spring Issues

ThePlay3r, 2025-04-27:
I have a brand new Moza R3 that I got for my Linux setup, using this driver and Boxflat.

When in-game, it works great, no issues so far.

However, when I get into a menu of some games, the wheel starts going to the right/left, same happens in some games when I go for example into the map, get out of the car, etc..

The issue is most prominent when loading into the game, for example in F1 2023 when I start the game (before driving) the wheel goes full force to the right and it keeps pushing until I load into the car.

I tried to set wheel spring both to 0% and 100%, only thing that changes is the strength of the feedback. (0% being the hardest)
I also tried setting "Game Spring" to 0% and 100%, this setting doesn't seem to do anything though.

When I get back "in-game"/to the car, it again starts behaving normally.

As you can probably tell, this is very frustrating and I can't really play any game like this, without the fear of damaging the wheel.

**System information:**
- OS: Linux Mint 21.3 Cinnamon (6.0.4)
- Linux Kernel: 6.8.0-58-generic (was on 5.15.0, but tried to update just in case)

Lawstorant, 2025-04-27:
I've noticed my R9 sometimes does that but usually a restart fixes this. I'll try looking into this, but it doesn't seem like it's an issue with the driver.

Some games try to center the wheel with constant force (so spring won't have any effect here and F1 is not using spring effect). They sometimes might have bad logic where they don't compute center but assume a value of 0 on the axis (as it normally should go from -255 to +256 for example). Some time ago Moza, for no apparent reason, changed the axis description and no it goes from 0 (full left) to 65535 (full right). I guess that's where this MIGHT be coming from.

I'll check F1 again. In the meantime, I think you could just disable the "wheel center" setting or make it 0%. It doesn't affect anything during race, just menu centering and maybe centering after car reset.

Lawstorant, 2025-04-28:
@ThePlay3r could you check if you have any FFB interpolation? Turn it down to 0 and test again.

I would honestly love to hide thos setting in Boxflat and force it to 0 bit some people might whine about it. This onterpolation setting is pretty bad and was causing my wheel to behave like that in DR 2.0 even in windows.

Lawstorant, 2025-05-05:
@ThePlay3r any updates on this?

ThePlay3r, 2025-05-05:
Yeah, sorry - didn't really had time to get to testing it out.

I've updated my whole system to latest version of Linux Mint, as you suggested, however that didn't fix the issue.

What did help, was indeed setting the FFB Interpolation to 0. (Default was 2)
(Boxflat in `Base` > `EQ` > `FFB Interpolation`)

Lawstorant, 2025-05-05:
@JacKeTUs can be closed

closed 2025-05-05 by JacKeTUs