JacKeTUs/universal-pidff #83: missing dependencies ? causes build failure?

Random-cheapskate, 2025-06-21:
I'm on the latest stable steam os build using kernel 6.11.11 

am having issues compiling the ffb driver .  here is a log dump .. im sure its probably my fault somewhere .

(deck@steamdeck universal-pidff-dkms-git)$ makepkg -scri --noconfirm
==> Making package: universal-pidff-dkms-git 1.0.0-2 (Sat 21 Jun 2025 06:26:00 PM EDT)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating universal-pidff git repo...
==> Validating source files with sha256sums...
    universal-pidff ... Skipped
==> Extracting sources...
  -> Creating working copy of universal-pidff git repo...
Cloning into 'universal-pidff'...
done.
==> Starting pkgver()...
==> Updated version: universal-pidff-dkms-git 0.1.0.r19.gbac70bd-1
==> WARNING: A package has already been built, installing existing package...
==> Installing package universal-pidff-dkms-git with pacman -U...
[sudo] password for deck: 
loading packages...
warning: universal-pidff-dkms-git-0.1.0.r19.gbac70bd-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) universal-pidff-dkms-git-0.1.0.r19.gbac70bd-1

Total Installed Size:  0.08 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                            [##########################################] 100%
(1/1) checking package integrity                                          [##########################################] 100%
(1/1) loading package files                                               [##########################################] 100%
(1/1) checking for file conflicts                                         [##########################################] 100%
(1/1) checking available disk space                                       [##########################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove --no-depmod universal-pidff/0.1.0.r19.gbac70bd -k 6.11.11-valve19-1-neptune-611-g88b36d49a5e3
:: Processing package changes...
(1/1) reinstalling universal-pidff-dkms-git                               [##########################################] 100%
:: Running post-transaction hooks...
(1/3) Arming ConditionNeedsUpdate...
(2/3) Install DKMS modules
==> dkms install --no-depmod universal-pidff/0.1.0.r19.gbac70bd -k 6.11.11-valve19-1-neptune-611-g88b36d49a5e3
==> depmod 6.11.11-valve19-1-neptune-611-g88b36d49a5e3
(3/3) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-neptune-611.preset: 'default'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-neptune-611 -g /boot/initramfs-linux-neptune-611.img
==> Using drop-in configuration file: '20-steamdeck.conf'
==> Starting build: '6.11.11-valve19-1-neptune-611-g88b36d49a5e3'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [steamos]
==> ERROR: binary not found: 'plymouth'
  -> Running build hook: [steam-deck]
==> WARNING: Possibly missing firmware for module: 'xhci_pci'
  -> Running build hook: [modconf]
  -> Running build hook: [keyboard]
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
==> WARNING: Possibly missing '/bin/bash' for script: /usr/bin/mount.steamos
  -> Running build hook: [resume]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-neptune-611.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
==> Building image from preset: /etc/mkinitcpio.d/linux-neptune-611.preset: 'fallback'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-neptune-611 -g /boot/initramfs-linux-neptune-611-fallback.img -S autodetect
==> Using drop-in configuration file: '20-steamdeck.conf'
==> Starting build: '6.11.11-valve19-1-neptune-611-g88b36d49a5e3'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [steamos]
==> ERROR: binary not found: 'plymouth'
  -> Running build hook: [steam-deck]
==> WARNING: Possibly missing firmware for module: 'xhci_pci'
  -> Running build hook: [modconf]
  -> Running build hook: [keyboard]
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
==> WARNING: Possibly missing '/bin/bash' for script: /usr/bin/mount.steamos
  -> Running build hook: [resume]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-neptune-611-fallback.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
==> Cleaning up...
(deck@steamdeck universal-pidff-dkms-git)$ cd

Lawstorant, 2025-06-28:
I don't really see any error related to `hid-universal-pidff` but you might wont to try the Steam Deck installation script. When I tried to just `makepkg` it before, there was missing stuff in the system and the Steam Deck script is reinstalling all the necessary stuff to make sure it will be there for the driver.

Lawstorant, 2025-07-03:
For the pedals/handbrake you can check out https://github.com/JacKeTUs/simracing-hwdb

Random-cheapskate, 2025-09-21:
> For the pedals/handbrake you can check out https://github.com/JacKeTUs/simracing-hwdb
> 
> If your devices aren't there, you could add them or at least provide us with VID:PID from `lsusb` in an issue. I guess your epdals/handbrake only have axes and no buttons?

i appreciate the heads up.  my apologies for not responding .  life has a way of messing with plans.

anyways.. im back and am at a total loss , i forgot the process to build and install these things on the steam deck .

may i ask for a little help ?   

thank fully after a few weeks of not touching anything my wheel still works , the pedals still only works on some games and the hand brake is still mia.