JacKeTUs/universal-pidff #90: Cammus C5 and "make debug" issues on bazzite

daemonoutcast, 2025-07-03:
Greetings, this is my first ever issue form so sorry for any formating errors. Attempted to make from source, returns this:

> Skipping BTF generation for hid_universal_pidff.ko due to unavailability of vmlinux

as such I cannot proceed to the last step of the manual installation.

dkms also wont work on bazzite.

Can provide more info if needed. Also worth mentioning the pedals and facebuttons arent detected but the base wheel appears on most programs.

JacKeTUs, 2025-07-04:
Hi! So you're on Bazzite? I thought Bazzite has one of the recent kernel versions... Is there any reason you don't want to use builtin one? What is your kernel version? 👀

daemonoutcast, 2025-07-04:
> Hi! So you're on Bazzite? I thought Bazzite has one of the recent kernel versions... Is there any reason you don't want to use builtin one? What is your kernel version? 👀

Hello. the built in package isnt allowing me to use the pedals for some reason. Connected separately or to the base. the face buttons ABCD work but nothing else aside from the steering axis.

Force Feedback settings on oversteer and on the android app work as far as i could tell. 

I also rebased from fedora silverblue and have the most recent kernel.

JacKeTUs, 2025-07-04:
Pedals isn't something that's covered with pidff or universal-pidff, pedals should be separate device. Manual installation from this repo sadly won't help you with pedals, as they are just not PID device.

Also i can't see any mention in Cammus user manuals that c5 actually having built-in usb hub for other devices. Hope i'm wrong, because there are photos with USB C ports on the back. 

I say, baseline should be - pedals connected to the PC directly. Can you share your kernel logs after connecting pedals to the PC? Is there something from `hid-generic`?

What DE are you using? If on KDE (Bazzite default one), check out Settings -> Game Controllers. Is the devices (pedals and wheelbase) both shows up there and you can see various inputs?

There is a chance that pedals (which are having 3 axis) are detected not as Joystick, but Accelerometer. If device doesn't show up in KDE menu, please try to install `hwdb` rules from here: https://github.com/JacKeTUs/simracing-hwdb. Follow instructions there, and after that, try to reconnect the device. 

About other face buttons - where did you test it? In game? From `wine control joy.cpl`? Or KDE Settings?

daemonoutcast, 2025-07-05:
Your hardware fix worked like a charm! Everything is detected now!

Thank you so much!

JacKeTUs, 2025-07-05:
Glad it's fixed now! I should probably list this repo somewhere where people will actually see it :D 

What about face buttons on the wheel though? If it was fixed too, i think it's because wheel was detected as gamepad, instead of joystick (yep, these are different things - think of a gamepad as xbox/ps ones)

daemonoutcast, 2025-07-05:
Yes, the facebuttons are also detected on all games. Thank you very much once again!

closed 2025-07-05 by JacKeTUs